profi200/open_agb_firm #1: Save detection

PickleRick-gbatemp, 2020-05-11:
> Save type hardcoded to 32 KiB (256k) SRAM. Unknown how save type detection will be done.

Why not use a native_firm application to select a rom and write the path (to rom and save) and the save type to "open_agb_firm.ini". Then trigger a reboot to open_agb_firm and it loads the ini file. I figure in native_firm there's more room to work with. You might even be able to write the config to ram since it isn't cleared on a reboot.

profi200, 2020-05-12:
That's not what i had in mind. The goal was to completely decouple this from userland (pesky GBA VC "injects" and stuff) and make it optionally configurable/bootable from userland (but it should itself have a menu for configuration/game selection). The main way of running this should be through some bootloader. Luma's chainloader or fastboot3DS for example.

Flash memory hates writes. I don't think writing to the SD card every boot is a good idea especially writing less than erase block size. And space is not a problem currently. I still have large chunks of VRAM free and AXIWRAM is basically barely used.

I have talked with endrift (mGBA dev) about save detection and i think we agreed the best way is having a game database with at least save type and hashes of known good dumps. But then the problem is building this database properly. So i will probably resort to string search-based detection in the meantime which is not accurate in some cases.

TriforceOfPwnage, 2020-05-12:
how would this game database for saves work with modified roms ie rom hacks?

profi200, 2020-05-13:
If the save type is the same and the game code matches the original ROM it will probably detect it fine however with a hash mismatch. When the game code is changed there will be no other way than save override.

TriforceOfPwnage, 2020-05-13:
by game code do you mean the game id? ie BPRE for fire red? if so that can be edited by hex

PickleRick-gbatemp, 2020-05-13:
Ah fair enough. 

> Flash memory hates writes. I don't think writing to the SD card every boot is a good idea especially writing less than erase block size.

That's why I also suggested writing it to an area of ram that doesn't get overwritten on reboot. That would be more ideal.

profi200, 2020-05-14:
Yes, i mean the 4 chars ID. It would be preferable if ROM hackers don't touch the ID. That would save everyone time.

Are there really cases where the ROM hack changes save type? Do they have a good reason to do that?

TriforceOfPwnage, 2020-05-14:
there are not many that change the save type. but there are many that add rtc to games that never have it ie. fired hacks.

profi200, 2020-05-14:
Ugh, which is basically a save type change. Not sure if i can get away with assuming these games use the RTC enabled save type.

TriforceOfPwnage, 2020-05-14:
wouldn't just editing the game id fix that? fore example bpre to axve. as that's ruby's id which uses a save type that's rtc enabled.

profi200, 2020-05-15:
I can't expect noobs to pull out a hex editor to fix this by themself. It just doesn't work.

TriforceOfPwnage, 2020-05-17:
tbh i think the people in the rom hack playing community should know how to hex edit if they go through the process of  of patching a rom with a computer much less installing a bootrom exploit on their 3ds to install cfw to run open agbfirm in the first place. and if necessary I can make a guide for you,for them, if it becomes a big enough problem.

profi200, 2020-05-17:
Remember that not only experienced people play ROM hacks. I don't see many issues regarding this coming yet. At least hopefully.

My motivation to do anything is kinda down right now. I'm trying to do something useful to get the project forward but it barely works :/

TriforceOfPwnage, 2020-05-17:
No problem i understand fully.

profi200, 2020-05-17:
I mean. It's already fully playable if we ignore the missing save detection and file browser. I completed Metroid Zero Mission with this a few days ago. But if you compile it with a RTC enabled save type beware that it doesn't set the RTC properly yet.

I'm thinking about splitting this further into the project and a "lib3ds" of sorts. I wanted to do this for a long time with the fastboot3DS source (this is based on it) since it has code for almost everything now.

TriforceOfPwnage, 2020-05-17:
Unfortunately i can't compile it myself because my pc is out of my hands for awhile. Btw are u saying you can compile it with a different hardcoded save type? I genuinely curious

PickleRick-gbatemp, 2020-05-17:
Might I recommend checking out the source code for Gericom's GBARunner2? It's a similar project to yours and he's already got save detection working. (He also has the 3DS gyro working in GBA games! That's cool.) His code runs directly on the ARM9 and ARM7 so I figured you might be able to reuse some of it.

TriforceOfPwnage, 2020-05-17:
actually gbarunner2 uses some kind of special patches in place of genuine save type support iirc.

PickleRick-gbatemp, 2020-05-17:
> EDIT: Yup uses special sram patches according to gericom [Gericom/GBARunner2#40](https://github.com/Gericom/GBARunner2/issues/40)

Well it can't patch everything to sram. Pokémon saves fine in it. If it was sram patched the save wouldn't be big enough and it would get corrupted.

TriforceOfPwnage, 2020-05-17:
if i remember correctly pokemon can run with sram saves abiet with a few issues that are not gamebreaking

PickleRick-gbatemp, 2020-05-17:
> if i remember correctly pokemon can run with sram saves abiet with a few issues that are not gamebreaking

No the save corrupts when you beat the elite four and it'll throw up an error about the save being damaged every time you start the game. GBARunner2 doesn't cause this behavior. See that linked commit.

TriforceOfPwnage, 2020-05-17:
yeah i saw the linked commit lol. that's why i deleted my comment.

profi200, 2020-05-17:
GBARunner2 works in a fundamentally different way to this. GBARunner2 runs in DS(i) mode and traps register accesses + emulating missing GBA hardware. This is basically a full GBA SoC implemented inside the 3DS SoC which means it has 100% compatibility (if we exclude GBA carts with extra hardware other than RTC and 64 MiB games).

The string search based save detection approach works but it's not 100% reliable which is why i wanted to have a database.

Save type can be changed here:
https://github.com/profi200/open_agb_firm/blob/master/source/arm9/hardware/lgy.c#L51
https://github.com/profi200/open_agb_firm/blob/master/source/arm9/hardware/lgy.c#L74

TriforceOfPwnage, 2020-05-17:
ok ive figured out how to change the save size on my own but what about the type?

profi200, 2020-05-17:
It's the GBA VC ROM footer values.

https://www.3dbrew.org/wiki/3DS_Virtual_Console#Config Below the table.

TriforceOfPwnage, 2020-05-17:
profii are you aware that the official register defs leaked recently?  according to gericom they make the agb save registers more clear.
Or is this old news to you lol

profi200, 2020-05-17:
I have them since over a year (stripped down comments but they were not so important anyway). If you look closely at some code you will see it's "heavily inspired" by official headers.

TriforceOfPwnage, 2020-05-18:
i played pokemon emerald Enhanced last night on open agb_firm and it ran flawlessly and so did the rtc.

profi200, 2020-05-19:
The RTC probably does run but it's reset to 1.1.2000 00:00 every boot. The biggest problem is probably that the 3DS side RTC is completely off unless you set it correctly with GodMode9. Other than that it's a matter of copying it from the 3DS RTC to the GBA one.

TriforceOfPwnage, 2020-05-22:
Profi is there somewhere else i can contact you? I got a few questions do you have a discord?

profi200, 2020-05-22:
No, but i'm on the #GodMode9 IRC channel on Freenode (which has a bridge to the GodMode9 Discord).

profi200, 2020-05-26:
In theory the RTC problem is solved now. Only day of week is not set but i don't know if this is used by any game.

TriforceOfPwnage, 2020-05-26:
That's great. Iirc tho pokemon does use day of the week in rom hacks and vanilla

TriforceOfPwnage, 2020-05-26:
@profi200 that's great. On the day of the week iirc the pokemon games use the day of the week in both vanilla and hacks

TriforceOfPwnage, 2020-05-26:
btw how would i change the save type to flash 1m with rtc now. it seems you changed the format

profi200, 2020-05-26:
I don't know if day of week is used as said. It's a bit annoying to calculate especially because it's encoded as value 0-6 where it's unknown which is Monday.

And nothing changed. I just use aliases (to keep it simple) for the same values:
https://github.com/profi200/open_agb_firm/blob/master/include/arm9/hardware/lgy.h#L12-L30

TriforceOfPwnage, 2020-05-26:
Well i know it uses the date for the berries when a berry is planted as the starting point. But i know for a fact rom hacks use the day of the week as you can change just the day of the week on your console and the events take place as scripted whether you go backwards or forwards in time

profi200, 2020-05-26:
"change just the day of the week on your console"? At least in Pokémon games you can't just change time/date. Did you confuse that with the DS games? And changing the day also changes the date. What i mean is a counter from 0-6 that tells you which day of the week it is. Not day of month (0-31).

TriforceOfPwnage, 2020-05-26:
what i mean is there is a tool that can DNS aka Days nights and seasons that actually patches the games to read the date of your device. so say your 3ds says its thursday of x week of it will load any event u code to only happen on such day in game. For example in fire red you could code let say an npc to only appear at night on a Thursday.

TriforceOfPwnage, 2020-05-26:
alright reviewing that source code of dns as far as i can see it seems it uses the date set at the start of the save to calculate day and time from then on every boot it offsets the time and day based on what i can see. it doesn't actually use the days of the week.

profi200, 2020-05-26:
I don't know what "dns" is. Never heard of it. The original games not using day of week is at least good news (however that's just Pokémon). So i assume it's working?

TriforceOfPwnage, 2020-05-27:
Dns stands for days nights and seasons its a binary change to the romhqc to use rtc when its base game  doesn't usually like fire red.

profi200, 2020-05-27:
I opened my original Emerald cart and it looks like it's using Macronix flash which is what i have been using in my test with this game. Theoretically the save type doesn't matter as long as the size and addressing match (and the game doesn't check the flash ID).
I found 1 little flaw with the current RTC handling. When starting a new game the Pokémon games reset the RTC to 1.1.2000 00:00:00 and if i save and reload the FIRM it jumps right back up to 2020. AgbBg solves this by using date/time offsets and saving them along the savegame. I can probably get away with this since mGBA does the same (just returns the current date/time).

[open_agb_firm.zip](https://github.com/profi200/open_agb_firm/files/4688212/open_agb_firm.zip)

TriforceOfPwnage, 2020-05-27:
thanks btw was looking at gbatek for the rtc based functionality of the gba and nds then found this

Day of Week Register (septenary counter) 
0-2 R/W Day of Week (00h..06h, custom assignment, usually 0=Monday?)
 3-7 - Not used (always zero)

seems t

Dartz150, 2020-05-27:
So I haven't found any other place to discuss this, so in the mean time Iwill use this issue post.
I have been testing this and so far is working to an usable state (I'm sure there are still many things to be added) and I'm pretty impressed on how it works far better than stock AGB_FIRM, congratullations!
I see that save types and saving are not implemented yet but while reading this topic it seems that these are very close to be added, so I will ask only two things to continue my tests (I'm rebuilding this each time you push new commits btw).

1.- rom.gba at the root is a temporary dir right? I supose a file explorer will be added later, or will it be fronted based much like how TwilightMenu works with nds-bootstrap?

2.- It is possible to enable the BIOS intro? I have also noticed that if no rom is loaded, the bios shows up like half minute later.

3.- I see that the color display is not washed out and a bit darkened, which looks excelent! Does it use a LUT? Or some kind of shadder?

4.- Is it planned to add a customizable layout for the screen stretch and palette darkness?

Thank you in advance, and very sorry if I bottered you with this questions, this looks pretty good so far and I would love to continue testing this alongside GBARunner2 which is another amazing project.

TriforceOfPwnage, 2020-05-27:
@Dartz150 hey dartz its Dualbladedkirito from dsi mode hacking discord! You can change the save type but you have to recompile it every time for now. Change it here https://github.com/profi200/open_agb_firm/blob/master/include/arm9/hardware/lgy.h#L12-L30

Dartz150, 2020-05-27:
Great, thanks for the tip, works as expected.
As suggested, I do agree that a database is used to determine the save type for the rom, and for romhacks, it should be fixed on the romhack side part as these aren't supported properly in AGB_FIRM as well, or maybe an external editable database could be used for that, so depending on the romhack, a new entry should be writen in plain text, to tell which save type is used with said romhack, specially on the RTC fire red romhacks.

TriforceOfPwnage, 2020-05-27:
Hey rtc function properly in rom hacks with this. Unlike regular agb_firm

Dartz150, 2020-05-27:
Great to know! Another point for this over AGB_FIRM, then I see that the main holdback now is how to standardize a proper way to detect the save type chip right? See my comment above.

TriforceOfPwnage, 2020-05-27:
Pokemon adventure red doesn't boot tho. Green bottom screen and black top screen

TriforceOfPwnage, 2020-05-27:
@Dartz150 also hes planning to make it have it's own rom select menu and be configurable within the firm iirc

profi200, 2020-05-27:
@Dartz150 
1. A file browser is planned.
2. Already implemented but hardcoded to off currently. Set this to true: https://github.com/profi200/open_agb_firm/blob/master/source/arm11/hardware/lgy.c#L39
3. It technically doesn't use a shader. The PICA200 supports no custom shaders. It's just this as used by mGBA: https://gist.github.com/profi200/e31df7d31b8c2cdccc0608aaa71681fc#file-main-c-L75-L78
4. Maybe later. Without a proper citro3d port this is not easy to make configurable at runtime. Color lookup tables like on original GBA VC can be added easily though.

@TriforceOfPwnage 
ROM too big/SD card corruption? This is the ARM9 calling panic(). ROM size is limited to 32 MiB (hardware limitation). I should probably implement proper errors.

TriforceOfPwnage, 2020-05-28:
is there any way to trim the rom down abit via hex? its 267 bytes over 32mib.

profi200, 2020-05-28:
Unless that data is not needed i'm afraid no. You could ask the autor of the ROM hack to try and cut the size down.

Games >32 MiB use a special mapper that is not supported by the cart emu: https://mgba.io/2015/10/20/dumping-the-undumped/

profi200, 2020-05-30:
The BIOS animation and save type settings have been moved here:
https://github.com/profi200/open_agb_firm/blob/master/source/arm11/main.c#L40

Save size is now automatically calculated based on save type.

Dartz150, 2020-06-01:
Mmm.. I'm trying to build the latest comit but I'm getting this error, it didn't happen before:

`make[2]: dma330as: No such file or directory
make[2]: *** [/c/open_agb_firm/arm11/Makefile:178: lgyfb_dma330.h] Error 127
make[1]: *** [Makefile:117: all] Error 2
make: *** [Makefile:60: release] Error 2`

profi200, 2020-06-01:
I didn't change anything related to makefiles. This tool must be somewhere in your path. Recommended way is to create some folder where you throw in all unofficial tools and add that to your path environment variable.

profi200, 2020-06-12:
Filesystem access has been reimplemented for ARM11 which i wanted to get done before i implement save type detection.

I will see if i can get the later working in the next days (it will be the inaccurate save type detection for now).

TriforceOfPwnage, 2020-06-14:
Do you mind sending me a build @profi200 with bios enabled to test

profi200, 2020-06-15:
Thought you can build this yourself now? Anyway.

[open_agb_firm.zip](https://github.com/profi200/open_agb_firm/files/4777466/open_agb_firm.zip)

profi200, 2020-06-16:
I need testers. The EEPROM save types are still mostly untested. Everything without the "Confirmed" comment needs testing. Unfortunately i don't know which games use these types:

https://github.com/profi200/open_agb_firm/blob/master/source/arm11/hardware/lgy.c#L104-L110

A test build from latest commit without BIOS animation to speed up testing:
[open_agb_firm.zip](https://github.com/profi200/open_agb_firm/files/4787415/open_agb_firm.zip)

[7 more replies]

closed 2020-06-23 by profi200