profi200/open_agb_firm #3: Classic NES series roms gives GAMEPAK ERROR

Dartz150, 2020-05-27:
See this for reference https://mgba.io//2014/12/28/classic-nes/

profi200, 2020-05-27:
Did you set the save type and size correctly? It's impossible for this to fail because it uses GBA hardware.

Dartz150, 2020-05-28:
I'm certain I did, I will test other Classic NES roms as I have been only using one.

profi200, 2020-06-06:
Got it to work? Also delete the save file when trying different save types.

profi200, 2020-06-08:
Closing for inactivity.

Dartz150, 2020-06-16:
Gonna bump this as I have made tests with the latest commit and the issue still persist, tried every available rom of the classic NES series from every region, all leading to the same "GAME PAK ERROR" issue. It may be related to an incorrect save type detection.

The save type detected on this type of games is 'EEPROM_V124', i tried to start the games with no save file and with previous save file created with mGBA.

profi200, 2020-06-16:
Did you make sure the dump is a known good one? https://datomatic.no-intro.org/?page=search

I'm gonna do some tests myself.

profi200, 2020-06-16:
I can confirm this but unsure why it's happening. Possibly the builtin hardware that "emulates" the save chip is not as accurate as it should. These games use various tricks most of which should be no problem at all to handle for the hardware GBA mode.

See: https://mgba.io/2014/12/28/classic-nes/

commit 837d3046 (2020-06-16): Properly detect Classic NES Series games save type. Fixes #3.
closed 2020-06-16 by profi200