profi200/open_agb_firm #6: Project name change

profi200, 2020-07-08:
The name of this project was the first that came to my mind when i had to decide but i'm not very good in giving names and i figured the name is not so easy to type so i decided it should be changed. However i need ideas for a proper name. Here are ideas i got so far from people on GodMode9 IRC/Discord:

openAgbFirm (it was a joke but might aswell include it)
YellowGba (CTR -> Citra -> Citrus -> yellow)

At the end i will probably make a poll with the most popular ones (if i get many, otherwise all).

chyyran, 2020-07-08:
*Agrume* is another word for citrus fruits and starts with 2 of the 3 letters of AGB_FIRM

Dartz150, 2020-07-08:
I have a few:

- CitrAGB
- CatGBA
- Lumalee (to acompany Luma and Rosalina as firms)

SleepyLark, 2020-07-08:
Here's my suggestion, if you want:
- Atlantis (The GBA's original codename)
- SeAGull_Boot (or just Seagull Boot or Seagull, basically play on of the letters "AGB", also another aquatic reference)
- Lime or LimeGBA  (or LemonGBA, or any fruit if you still wanted to stick with the citrus reference)

PepeShake, 2020-07-08:
Serious response: FrankenAGB
Because it's homebrew using the built in GBA hardware. 
Meme response: AyyGeeBee
Because it's dumb.

piratesephiroth, 2020-07-09:
Isn't it an open source replacement for the stock AGB_FIRM?
If it is, then Open AGB Firm is great..

DullPointer, 2020-07-09:
KrackedGBA a.k.a KGBA
GameBoy Advanced Evolved a.ka. GBÆ

profi200, 2020-07-09:
Sounds almost like _EA_. Pay $1 to unlock this game :p

> Isn't it an open source replacement for the stock AGB_FIRM?
> If it is, then Open AGB Firm is great..

Well yeah, but it's a boring name i think. There is nothing original to it.

> I have a few:
> 
>     * CitrAGB
> 

I think that one is already taken. https://github.com/shinyquagsire23/gpsp he named it CitrAGB (but the repo has a different name for some reason).

derrekr, 2020-07-09:
RealGBA,
GBA²,
gba43ds/gba4ctr

vaguerant, 2020-07-13:
> YellowGba (CTR -> Citra -> Citrus -> yellow)

Along these same lines, FreshGBA. It's lemony fresh, and it is new (fresh) as well.

Lv44EnderShaman, 2020-07-22:
How about continuing the names based on citrus types, like Citra and Yuzu for the PC emulators? Those are cultivars of citrus fruits. There's a range of variety you could use. Examples include 'Pomelo', 'Papeda', 'Amanatsu', 'Oroblanco', 'Limetta', or 'Lumia'... Just throwing that out there ;)

profi200, 2020-07-22:
That's a possibility.

I will probably soon™ start the poll since there were no new ideas for a while.

cheatfreak47, 2020-07-29:
How about **Gasoline** or GasAGB for short
idea being it goes backward from Nitro, which was the DS

gabrielmathus, 2020-08-01:
GBAFirm could be something xD

PepeShake, 2020-08-01:
> How about **Gasoline** or GasAGB for short
> idea being it goes backward from Nitro, which was the DS

Oh I like that. Seconded.

lagseeing, 2020-09-01:
GuavaGBA. Guavas have more Vitamin C than oranges, and Vitamin C is a thing most citrus fruits have in common

ergoithz, 2020-09-01:
AGBmode anyone?

TheGinGear, 2020-09-01:
AGBoot
3DBoy 
SimpleAGB
InstaGB
GBArm9
StellarAGB or StellarGBA, (cause it's outside of the Horizon? Eh?)
GBAstronaut could also work with the horizon pun.

Idk, you could really go somewhere with the space theme/past the horizon pun.
CometBoy?

missionfloyd, 2020-09-02:
The name is fine how it is. Everyone insists on coming up with ambiguous names for their projects.

If you didn't know any better, what would you assume 'vim' or 'nano' were? But 'Notepad' is perfectly clear, isn't it?

TheGinGear, 2020-09-02:
Even though I provided a bunch of names, I wanna make it clear that I agree with @missionfloyd ; The name isn't bad or anything, and it's both descriptive and memorable. It might have needed a custom name if it wasn't a piece of software that basically has no other program to differentiate from, like the vast majority of windows programs. But with how it's genuinely the only thing you'd find of you search for an open-source agb firm, it's unique enough to have an identity already.

profi200, 2020-09-06:
Well, since the "keep it" option is popular i should maybe first make a poll for change it or keep it (but removing the underscores maybe).

Someone knows a poll site that doesn't try to collect every data it can?
(Looked at strawpoll but they collect and sell data clearly mentioned in the privacy policy.)

lex3a, 2020-09-13:
> Well, since the "keep it" option is popular i should maybe first make a poll for change it or keep it (but removing the underscores maybe).
>
> Someone knows a poll site that doesn't try to collect [...]

Yeah, just leave it as it is but without underscores. Or make an accent on the extension -OpenAGB.FIRM

twocows, 2020-09-21:
I'd say either keep it or at least keep it simple. GBARunner2 has a good name: it says what it does in a way a casual user can understand. I'd vote for something like GBAfirm or even something like NativeGBA or GBA3DS.

HTV04, 2021-02-20:
I think "GBAFirm" would be a good name. It's unique and it gets to the point, it's a 3DS firmware focused on using the internal GBA hardware.

Filyx20, 2023-01-09:
With the likes of godmode9,  b9s a9lh and since the gba runs under the arm9 cpu, having 9 somewhere in the name would be fun to have

fox8091, 2023-01-11:
@Filyx20 The GBA doesn't run under the ARM9 CPU though? It uses the ARM7?

Masamune3210, 2023-01-11:
correct, the firm runs on the arm9 but the majority of the work happens on the arm7

profi200, 2023-01-11:
Yes and no. The game runs normally on ARM7 inside the hardware sandbox. ARM9 is mostly doing nothing except when ARM11 tells it to backup the savegame.

karasuhebi, 2024-11-26:
I agree with keeping the current name. I never had any problems with it and as other people have mentioned, it's one of those names where "it does exactly what it says on the tin"

But if you're still looking for name suggestions, something with the word "ambassador" in it would be good because I think using that word would make it easier for people who don't know what agb_firm is to quickly understand what this does since it references the fact that you're using the same method the 3DS Ambassador Program uses for GBA games. So maybe something like openAmbassador? Or openGameBoyAmbassador to where your abbreviation could be oGBA or openGBA?