profi200/open_agb_firm #7: Save detection fails on Pinball Challenge Deluxe (Europe).

vaguerant, 2020-07-14:
This game has saving support but doesn't include any of the typical save type strings, so it would need to be hardcoded as an exception. The publisher Ubisoft decided to pull the battery backup before release, presumably as a cost-cutting measure, but saving support was left in the game and I can confirm saving works on real GBA hardware with a flash cart.

With apologies, I have no idea what save type the game expects, so I'm not any help in terms of getting this working.

profi200, 2020-07-14:
It can't be auto detected. Can only add it to the override list. If the devs intended it not having a EEPROM/Flash/SRAM chip it would be better making that optional imho.

vaguerant, 2020-07-15:
Just for reference here's a thread about this game on the old Pocket Heaven homebrew message board:

http://archive.is/bAQiB

It includes a quote from Binary9's director of development, Stewart Gilray, where he talks about the game being designed with saving in mind.

The thread also has some discussion about what save type we think the game uses; it seems it may actually be SRAM rather than EEPROM since it works on SRAM-only flash carts without any form of patching.