profi200/open_agb_firm #8: Loading from fastboot3DS v1.2 menu (not keycombo) results in screens being turned off

GerbilSoft, 2020-07-28:
Attempting to load open\_agb\_firm from the fastboot3DS v1.2 menu results in the screens being turned off. Presumably it's running, since pressing B shuts the system off, but with the screens off it's rather difficult to do anything.

This does *not* happen if open\_agb\_firm is assigned to a key combo and the key combo is used to boot open_agb_firm. It also doesn't happen if loaded through GodMode9 v1.9.1.

profi200, 2020-07-28:
Known problem that also happens with Luma and b9s (iirc). It has been fixed by porting the proper screen init code to fastboot3DS. Unsure when but a (fb3DS) release is coming. Until then the fix is already in master in case you can live with recompiling it. Unknown when this will be fixed in Luma.

The problem is it is triggering a panic() but you can't see it with screens off. This is because it receives MCU events it should not be receiving due to bad screen init code in the bootloader/payload loader (whatever you wanna call it).

GerbilSoft, 2020-07-28:
Makes sense. I'll take a look at building the latest fastboot3DS then.

closed 2020-07-28 by GerbilSoft