profi200/open_agb_firm #10: Only boots from Luma3DS chainloader with more than one firm.

PepeShake, 2020-08-10:
If you try to boot this from Luma with only OAF in the payload folder just shuts off your 3DS. In fastboot3ds trying to boot it with hotkey also shuts off the system but you attempt to boot it from the menu it locks up. The only way I've gotten it to boot is with multiple firms and the Luma chainloader.

jonirrings, 2020-08-10:
@PepeShake You need a recent build of fastboot3ds, I've built [one](https://static.jonirrings.com/fastboot3DS.firm) some days ago. 
And it works fine on my N3DS.
You could build it yourself, and install it via GodMode9.

how to install fastboot3ds via GodMode9
1. boot into GodMode9
1. navigate to you fastboot3ds.firm
1. press A on fastboot3ds.firm
1. install as firm0

be careful, this operation may damage your 3DS

profi200, 2020-08-12:
It's a known problem caused by receiving MCU events (from the bootloaders screen init) it should not be receiving which triggers a panic() (can't see it with screens off but pressing a button powers off). I have done everything i can from my side and fixed my screen init. The only thing i can do is delay the screen init significantly which increases startup time noticeably. Not really something i want.

In fastboot3DS this has been fixed but there is no new release yet. We will work on that soon. In the meantime use the workarounds mentioned.

profi200, 2020-10-29:
Forgot to mention this here. Try the latest alpha build if you still have this problem. It should work with Luma and fastboot3DS 1.2.