profi200/open_agb_firm #11: Headphones don't work

BrittanyOfKoppai, 2020-08-11:
When booting open_agb_firm (I used the no intro version) despite my headphones being plugged in, the audio still streamed through the system's built in speakers.

profi200, 2020-08-12:
I don't know why the automatic output switching is broken for a select few people but i know how Nintendo handles this and gets around the issue.
That said it's not high priority but i will look into it eventually.

PepeShake, 2020-08-13:
I get this problem sometimes. Unplugging the headphones and plugging them back in seems to fix it.

wiceli, 2021-02-16:
I have the same problem, the sound doesn't come out from the earbuds! the latest release is used.

profi200, 2021-02-17:
As mentioned above i don't know why the auto detection is broken but it can be manually handled by the CPU.

sak-pase, 2021-09-27:
I only have this problem with Apple headphones, and I can usually get them to work by holding the Play/Pause button on them for about a second after re-plugging them in. Probably has something to do with Apple's disdain for industry standards.

profi200, 2021-09-27:
Not sure about that. I still don't know why the output switching is so wonky. Going by a similar datasheet it measures output pins and switches if it detects changes. In theory it's fine. No idea why it's broken in practice. I still have not implemented the second detection method Nintendo uses.

OrangeFlavored, 2021-11-07:
Seems like it's dependent on the cable used or something, using M-M extension cord into portable amp which my headphones are connected to was not detected, but plugging the same headphones directly into the 3DS worked fine.

profi200, 2021-11-07:
Forgot who posted it originally but it seems the issue is exclusive to 4 pin headphones/headsets. You can usually make them work by holding a button for 1-2 seconds (tested it). If there is no button that won't work of course.

PiptonyZX, 2022-09-27:
Used 3 pins for short range cord and it work and no button at all.   It work fine.  I tested long range, didnt work.   Only work under 3ft cord.

PiptonyZX, 2022-09-27:
Forgot to mention, only one speaker does work.   Model name is 19YX01.  Brand name, well it kinda weird name and it is pretty long.  Some kind of language I think.

SHENZHEN JIA HUA LI DIAN ZI YOU XIAN GONG SI.

If you want use other speaker, then you must leave cord on your 3ds.  If you take it off on 3ds, then it wont work wne you plug it up.   You have to start over from gm3 menu.  (Not main menu where you dump it) I test it with other speaker to plug and it work as long you jeep cord on your 3ds.   If you leave cord on speaker and 3ds after turn it off, then you wont have to use this model of speaker again as long you dont take iff from 3ds.  If you do, then you have to use this model of speaker.

profi200, 2022-09-27:
Might have mentioned it above already but it's not clear to me how CODEC does the detection. Might be measuring impedance or similar. And as mentioned Nintendo uses a different method of detection that uses a physical switch inside the headphone jack. I will implement Nintendos way of detection to fix these issues eventually.

PiptonyZX, 2022-09-27:
Here is video to proof.

Oh okay.   Hope work it out with Nintendo.

https://youtu.be/3IQiNLMxsu4

profi200, 2023-08-05:
This is now fixed. See nightlies/actions tab.

closed 2023-08-05 by profi200