profi200/open_agb_firm #12: Implement a method to set scaler to 1:1

uncon, 2020-08-19:
I believe there should be some way to use 1:1 video scaling. (Personally, I do not like the smeared look of non-integer scaling and am willing to sacrifice screen real estate for sharper pixels.)

Since there is currently no configuration UI, this could be implemented as a separate build, but I understand this would be _another_ build and added effort for binary releases. So, this is an enhancement request that would be nice-to-have.

profi200, 2020-08-20:
I wanted to look into it but ran into a problem. The matrix scaling hardware is currently not only used for scaling but also to fix color accuracy. And this doesn't seem to work for 1:1. I probably have to resort to using the GPU.

It's also getting kinda messy with hardcoded GPU command lists. I will probably not be able to offer proper runtime configuration of scaling until i have a working citro3d port.

stranno, 2021-01-14:
Hey @profi200, is the 1:1 scaler fully implemented now?

I have tried the latest commit and display seems to be completely messed up using SCALE 0.

profi200, 2021-01-14:
It is, but it's not a simple one line edit. The hardcoded GPU cmd lists and the hardcoded DMA code is what prevents me from adding a scale option to the config. It's kinda annoying not having a citro3d port and a DMA JIT to get rid of that problem. I will see if i can somehow make this work via a compile time define.

stranno, 2021-01-14:
Oh, ok. I though it was that simple x-)

Good luck figuring it out.

profi200, 2021-01-14:
For the time being:
[open_agb_firm.7z.zip](https://github.com/profi200/open_agb_firm/files/5816762/open_agb_firm.7z.zip)

Rename ".7z.zip" to ".7z". Github won't allow 7-Zip archives in issues for some fucked up reason.

stranno, 2021-01-14:
Thank you! It works flawless.

blackbeary, 2022-02-28:
> For the time being: [open_agb_firm.7z.zip](https://github.com/profi200/open_agb_firm/files/5816762/open_agb_firm.7z.zip)
> 
> Rename ".7z.zip" to ".7z". Github won't allow 7-Zip archives in issues for some fucked up reason.

Just want to say thank you for compiling this, it's the only way I can play the mother 3 english
translation without ANY lag in my favorite scaling !! with headphones nonetheless, very thankful !

MitchTalmadge, 2022-05-02:
For anyone coming across this today, there is an option called `scaler`, which if set to 0, will result in 1x scale on the top screen. See https://github.com/profi200/open_agb_firm#video. It seems that there were problems with it before, but it's working great for me now.

profi200, 2022-05-02:
I think this can be closed now. 1:1 scaling has been available for a while.

closed 2022-05-02 by profi200