profi200/open_agb_firm #13: Uploading GBA roms remotely?

exelotl, 2020-08-25:
Hi!

I've been wondering if it would be possible to receive and immediately launch a GBA rom over my local network? E.g. using a simple protocol like [3dslink](https://github.com/devkitPro/3dslink) as supported by the standard hbmenu.

As a GBA homebrew developer, I could imagine this would become the fastest and most ideal way to test my game on real hardware!

Many thanks for your amazing work thus far :)

profi200, 2020-08-26:
Unfortunately the answer is not any time soon. Implementing a WiFi module driver + network stack is very complex and time intensive. The only alternative i could try is via infrared but i neither have a USB dongle nor any guarantee this would work.

exelotl, 2021-03-04:
So I ended up purchasing a WiFi-enabled SD card (Toshiba FlashAir), which is working an absolute treat with my Old 2DS!

I was able to put it into "station" mode (i.e. it joins my local network instead of acting as a hotspot) and use it as a network drive via WebDAV, by making some changes to the config file on the card.

* [Quick tutorial](https://mattshub.com/blogs/blog/flashair-sd-card) (Matt's Hub)
* [FlashAir Developer Docs](https://web.archive.org/web/20190916221054/https://flashair-developers.com/en/documents/api/config/) (wayback machine)

I was honestly amazed to learn that a device like this even exists. There are a handful of other wireless cards on the market which may also be suitable, but this one seems to be the most convenient and configurable by far.

New 3DS/2DS/XL users are unfortunately out of luck because there's no such thing as a WiFi enabled _Micro SD_  card. I guess we can dream...

profi200, 2021-03-04:
I have one of these from Transcend but i never really used it. They take a while to connect and they hog power like crazy. Another way to upload the ROM could be through the gamecard slot. In fact i'm working on doing this for FIRM files already. Requires opening a DS game, removing the EEPROM/flash and soldering wires to the pads for SPI transfer.

joshbackstein, 2023-04-05:
@exelotl I know this is an old issue, but I created a 3DS homebrew app that basically does this. You run the homebrew and it waits to receive a ROM over the network, then it saves it to the SD card and reboots into open_agb_firm.

Repo: https://github.com/joshbackstein/gba-net-boot

With Luma3DS v12.0 and later, you can set it up to boot straight to hbmenu, and if you replace boot.3dsx, you can even boot straight into the loader.