profi200/open_agb_firm #14: Doesn't boot

currall, 2020-08-31:
I used the latest .firm from the releases page and it just has a black screen. The power light is on but no other light.

Is there a certain place I should put the .firm or the roms? At the minute my firm is in the payloads folder

profi200, 2020-08-31:
Probably you are running into screen init incompatibility. Until Luma is getting updated with the fixed screen init there is not much you can do besides having multiple FIRM files in the payloads folder and selecting open_agb_firm from there.

See issue #10.

closed 2020-08-31 by profi200

currall, 2020-08-31:
I use fastboot3ds not Luma chain loader. I boot from the .firm from there.

profi200, 2020-08-31:
See my answer in the linked issue. It's the same with the current fastboot3DS release. Has been fixed in master but there is no new release yet.