profi200/open_agb_firm #16: Implement ROM mirroring

Powerlated, 2020-09-20:
[image]
Currently, ROM out of bounds is padded with 0xFF.

profi200, 2020-09-20:
That picture is a redirect to this issue.

If you are referring to mirrors of the ROM area (0x08000000, 0x0A000000, 0x0C000000) this is unfixable and how the hardware is implemented.

profi200, 2020-09-23:
I asked endrift about this and the only games which actually do mirror the ROM are the Classic NES series which do use this as AP (but strangely they work fine even with 0xFF padding). Every other game doesn't respond to out of bounds reads resulting in getting "open bus" data back. This is something i can't fix because the hardware gamecart emulation uses fixed size 16 and 32 MiB and there is no way i can reduce that to mimic open bus behavior.

The failing SRAM tests i can't fix either. Seems like a hardware bug for unaligned reads/writes. The H-Blank test failing is also failing on a GBA SP with the exact same result (one off) so i think that test should actually be a pass.

closed 2020-09-23 by profi200