profi200/open_agb_firm #17: First launch works, next launch crashes (hangs)

shar33f, 2020-09-20:
I'm running open_agb_firm alpha build 2020-09-08 on a new 3ds (non-xl) and the first time I launch it, it works perfectly; I see the file browser on the lower screen and can load any rom no problem (it runs really nicely!!). When I'm done playing, I press the power button to turn off my 3ds and then press power again while holding start and both screens just stay black and the blue power LED stays on solid. I have to pull the battery, and on next try it works fine again. I have to reset my system date and time every time I pull the battery, otherwise everything works fine but would be nice if it worked every time. Is anyone else experiencing the same issue?

profi200, 2020-09-20:
Does it power off if you press A while on the blackscreens? If so this is the known screen init incompatibility between release fastboot3DS/Luma/GodMode9.

shar33f, 2020-09-20:
Yes that works, thank you! So much better then pulling the battery and then having to reset the date and time :) I'll use this workaround and wait patiently then until this is addressed in a future release. Until then thanks so much for this amazing software!!

profi200, 2020-09-20:
Duplicate of issue #10.

closed 2020-09-20 by profi200