profi200/open_agb_firm #18: Issues booting open_agb_firm twice in a row

twocows, 2020-09-22:
If I boot open_agb_firm and run a game, then power off, then try to boot into open_agb_firm again, it just hangs. If I then boot into the normal system software, let it come up, power down, then boot into open_agb_firm again, it works fine (but the problem will still come back if I try to do it twice). I didn't have this problem before updating to build 2020-09-08.

I'm using an O3DS XL, Luma should be on the latest version. This is using the version of open_agb_firm that has the GBA bios splash.

shar33f, 2020-09-22:
It's a known bug, see https://github.com/profi200/open_agb_firm/issues/17 and https://github.com/profi200/open_agb_firm/issues/10

Hopefully will be fixed soon in a future release.

profi200, 2020-09-22:
Duplicate of #10.

closed 2020-09-22 by profi200