profi200/open_agb_firm #21: Implement change ROM and last rom folder

redirectto, 2020-10-22:
Is it possible for a newer version to be able to press a combination of buttons to select another game instead of having to turn off the console, also it would be nice if open_agb_firm remembered where it was the last folder you loaded a game

profi200, 2020-10-22:
It's not possible to load a different ROM once GBA mode is running. It requires a reboot/poweroff. Remembering the last dir a ROM got loaded from is something i can add.

foxsouns, 2021-05-03:
This issue should probably be closed: everything's been resolved.

profi200, 2021-05-03:
Yeah, it didn't auto close yet because the commit is not in master yet.

Dimensional, 2021-09-08:
Would a reboot that automatically takes you back into open_agb_firm be a plausible solution? No having to hold down start for Luma, or a button combination for FastBoot3DS?

profi200, 2021-09-09:
That's pretty much the only way. Need to see how this can be done. But it's kinda low priority.

commit bbcfb69a (2020-10-24): Added game database search code (and a debug menu for it).
Mimic ROM "open bus" (not 100% accurate) and ROM mirroring. Fixes #21.
Implemented some useful ARM instruction intrinsics.
commit c87769a0 (2020-10-25): Added returning to the last dir a ROM was launched from on next boot. Fixes #21.
commit ff8ebb9f (2021-11-07): Added returning to the last dir a ROM was launched from on next boot. Fixes #21.
closed 2021-11-07 by profi200