profi200/open_agb_firm #23: Request: Display version / build information in file browser screen

shar33f, 2020-11-03:
Just wondering if it would it be possible to display the build or version information of open_agb_firm somewhere on the file browser screen so it's easy to know what version is running (and less confusing when upgrading). Thanks!

profi200, 2020-11-03:
Wanted to do that for a while. Right now i'm busy enough to find a very annoying and hard to find bug that is blocking the entire development.

shar33f, 2020-11-03:
All good, sounds like you have bigger fish to fry! Wishing you good luck to squish that bug!

karasuhebi, 2024-11-26:
Not sure if this has been implemented yet since I haven't updated in a few years but if it hasn't, I would like to second this request. :)