profi200/open_agb_firm #24: Request: Fast scroll in file browser

shar33f, 2020-11-08:
I know you have way bigger fish to fry, but hoping you can squeeze in hold-to-scroll in the file browser in a future release. It would make locating roms within large directories so much easier. And maybe the option to press dpad right and left to page up and down? Thanks again so much for the amazing work you are doing!

profi200, 2020-11-08:
Left and right has been in pretty much since the first file browser build. The file browser is very bare bones since i don't want to spend too much time on it. It will be replaced in the future but i'm undecided what to use for the GUI right now.

shar33f, 2020-11-09:
Oh man I didn't even try left and right, thanks for pointing that out!! Cheers