profi200/open_agb_firm #25: Button remapping

TheRozitaKnight, 2020-11-18:
Using the X/Y buttons would be really useful in certain games. I tried remapping buttons by modifying the source code and compiling it, but it only affected the file browser. Is it possible to implement button remapping?

profi200, 2020-11-19:
Button remapping is not implemented. Buttons are directly wired to the GBA hardware internally however there are override regs. The button override regs would need to be updated each VBlank which might be too late and the input is delayed by one frame (coming on top of the unavoidable >1 frame output delay). Not sure if this is desirable.

seriouslyunderappreciated, 2020-12-01:
Perhaps this can be implemented as a separate distribution? There are games where I would gladly eat some input delay in exchange for remapped controls.

seascape, 2023-03-03:
Just as a little more context for why this would be desirable, I found my way to this issue because in official Nintendo VC emulations like on Switch and 3DS I nearly always want to remap B-A to the much more ergonomic Y-B, and hoped that option would be feasible in open_agb_firm.

Very cool project either way though.

commit 014c341e (2023-08-06): Added button remapping. Closes #25.
closed 2023-08-06 by profi200