profi200/open_agb_firm #26: Folder names beginning with an underscore disrupt alphabetical organization.

dwain12435, 2020-12-13:
Title is self-explanatory.

dwain12435, 2020-12-13:
Also, why is /3ds/open_agb_firm created if nothing is stored in it?

dwain12435, 2020-12-13:
Also, please either remove texture dumping or make a seperate distro that removes texture dumping because it is annoying for the game to freeze every time you accidentally press Y.

profi200, 2020-12-13:
Folder contents are not naturally sorted is why it's not quite right. For the time being i won't change that but i will once i implement a proper GUI.

"/3ds/open_agb_firm" is not unused. "lastdir.bin" is written there and recent commits store config.ini there.

The freezing is actually not intended. It's a bug i was not able to track down yet. Pressing HOME can unfreeze in most cases. I will probably change the screenshot/texture dumping to use 2 entirely different buttons to make it harder to trigger.

dwain12435, 2020-12-15:
OK, can you make it where it doesn't make SD:/3ds/open_agb_firm if it
doesn't store anything in it? (I launch all my games from SD root, so no
lastdir.bin, and it makes the empty folder anyway.)

profi200, 2020-12-15:
It will be required anyway soon when i make the next alpha for the config file and maybe in the coming months for the game save database. Also screenshots will go into "/3ds/open_agb_firm/screenshots" when i implemented that (depends on mentioned freeze bug).