profi200/open_agb_firm #28: Multiupdater compatability

willis936, 2021-01-07:
Could the release format support MultiUpdater?  MultiUpdater does not work with 7z, only zip.  Switching to zip or having the firm file not be in an archive would fix this.

https://github.com/LiquidFenrir/MultiUpdater

willis936, 2021-01-07:
Nevermind, 7z works.  This was my mistaken.

This MultiUpdater config json entry works.

```
{

    "name": "open_agb_firm (latest release)",

    "url": "https://github.com/profi200/open_agb_firm",

    "inrelease": "open_agb_firm.*\\.7z",

    "inarchive": "open_agb_firm\\.firm",

    "path": "/luma/payloads/open_agb_firm.firm"

}
```

closed 2021-01-07 by willis936

profi200, 2021-01-07:
Probably the same issue as mentioned here: https://github.com/Universal-Team/Universal-Updater/releases