profi200/open_agb_firm #29: [Request] Filter options

Kirit29, 2021-01-14:
Is it a possibility to add filter options editable in the configuration? I'm one of the few who actually like full screen gba with bilinear filtering XP

profi200, 2021-01-15:
Not "full" screen but aspect ratio fit. Setting -DOAF_SCALE_SELECT=1 should use bilinear scaling.

https://github.com/profi200/open_agb_firm/blob/kernel_experiments/arm11/Makefile#L27

Kirit29, 2021-01-15:
Any way you could compile it for me? I don't have access to a pc and I would need to root my tablet if I wanted to do it on my tablet.

profi200, 2021-01-15:
I can't make this configurable for now. That needs a citro3d port.

[open_agb_firm_bilinear.7z.zip](https://github.com/profi200/open_agb_firm/files/5821308/open_agb_firm_bilinear.7z.zip)
Rename ".7z.zip" to ".7z".