profi200/open_agb_firm #30: [Request] bottom screen with ratio fit

ken-likun, 2021-01-27:
Is it possible of using bottom screen to play, meanwhile ratio fit.
Ratio fitting at bottom screen is ture fitting.

profi200, 2021-01-27:
Possible? Yes, but the scaling is even worse at 1.333333333x compared with 1.5x. I'm unsure if the scaling matrix can even do this uneven of a scaling factor. Bilinear is no problem of course.

But unfortunately i can't make the screen or scaling runtime configurable yet. I need a citro3d port for that and a DMA-330 JIT so i can change the DMA transfers at runtime.

ken-likun, 2021-01-27:
I mean is it simple of displaying at bottom screen using 1:1 video scaling. Personally I think 1:1 video scaling of bottom screen is more like GBA's display.
If it is simple, could you compile it for me? This is an enhancement request that would be nice-to-have.

Powerlated, 2021-02-16:
Good idea. Would also open up the possibility of making GBA Macro mods with the 3DS.

DragonWario, 2024-05-21:
The 3DS's bottom screen uses standard square pixels too, so it may be a little sharper for videophiles. Top screen uses a Pixel Aspect Ratio of 1:2, resulting in the need scale 1 standard pixel to 2 horizontally stretched pixels; resulting in and a blurrier image. There is a way to unpair the pixels and have the game only, such as use half of the "paired" 400px width (the effect used on SNES9x on 3DS) but I would expect there is no way to activate that mode in GBA Firmware: Which is why the Bottom screen may be the only "realistic" option.

profi200, 2024-05-22:
It's very much possible to use wide mode (800 pixels) if that's what you meant but it's more difficult to render. Have not been getting around to it. The pixels are no longer square in this mode and the GPU is the only hardware that can scale this.

The bottom LCD doesn't really look any better despite having square pixels. It's rotated just like the topscreen which makes it look bad too.

D-Maxwell, 2024-07-02:
It may look just as bad, but I for one sure would be glad were I able to play GBA games on the bottom screen of my 3DS ; and it would not be "even" in 1:1 scaling, but "hopefully" in that aspect ratio. Non integer scaling is unnecessary and overly complex for the little it achieves.

Would the bare bones functionality of having a boolean toggle in the settings for which screen to display onto possibly be an idea that you may consider for a later release ?