profi200/open_agb_firm #32: No maximum backlight brightness option

HTV04, 2021-02-19:
I don't know if this is just me or not, but the backlight brightness, even at 64, seems a little too dark. Is the limit really 64? I tried checking GodMode9's code and I think it had a limit of 210.

Also, a bit unrelated, but what's the best outGamma setting for "accurate" colors? I set mine to 2.2, same as the inGamma setting, and that seems to work well. Don't know if this is the right gamma to set it to, though.

profi200, 2021-02-19:
64 is the limit for now. Until brightness control is fully implemented i can't allow more because inevitably someone will destroy their backlight with too high values.
There are various per device (or panel) PWM settings and oaf only uses the default right now which makes the full range available for older models but not necessarily the newer ones.

The default outGamma is basically as close to raw output as possible. The LCDs are not calibrated at all out of the factory and have no setting for this either. The default tries to mimic the identity matrix output of GBI:
https://gc-forever.com/wiki/index.php?title=Game_Boy_Interface/Standard_Edition#Color_matrix

Having an option for GBA LCD color emulation is planned but i can't make any promises because it involves matrix math the GPU can't do accurately due to missing fragment shaders.

HTV04, 2021-02-19:
That sucks, but I understand. I wish it were just a little brighter though, the current brightness limit is less than half of what the brightness on my 3DS usually is.

As for the default 1.54 gamma setting, I think it looks a little off, especially on my New 3DS. What's the gamma setting that best matches GBA emulators?

profi200, 2021-02-19:
It looks off because the LCD panels are trash. Even if you correct gamma it still looks off. https://www.neogaf.com/threads/detailed-review-of-switch-screen-quality-and-compared-to-3ds-by-erica-griffin.1369087/

Emulators output raw unless you use filters.

HTV04, 2021-02-19:
Oh. I guess it isn't possible to output raw?

profi200, 2021-02-19:
inGamma = outGamma is basically raw output. It looks very washed out due to the bad LCD panels.

HTV04, 2021-02-20:
Okay, thanks. Anyways, I think I'll close this issue, looking forward to the backlight control feature!

closed 2021-02-20 by HTV04