profi200/open_agb_firm #34: Sonic The Hedgehog - Genesis not saving

HTV04, 2021-02-20:
I know, it's an absolutely horrible port, but bear with me.

*Sonic The Hedgehog - Genesis* doesn't seem to be able to save. According to No-Intro, it uses EEPROM_v124, like *Mario & Luigi: Superstar Saga* does.

Upon booting the game, the effects of the game being unable to save start to appear. The music and sound options are set to off, and the game has a "Resume" button that isn't supposed to appear yet.

When playing the game, the "Saving..." pop-up still appears as if saving were working, but it's probably in vain, because after I turned the game off and checked my SD card, there was no .sav file.

I guess this has something to do with the buggy EEPROM detection. Can this be fixed at all before a save type database is implemented?

Speaking of save type databases, I believe [VBA-M has one](https://github.com/visualboyadvance-m/visualboyadvance-m/blob/master/src/vba-over.ini). Not sure about mGBA though.

profi200, 2021-02-20:
You can add an override here and compile it:
https://github.com/profi200/open_agb_firm/blob/kernel_experiments/source/arm11/open_agb_firm.c#L211-L215

HTV04, 2021-02-20:
Sounds good. I'll just encountered a similar issue with Mega Man & Bass, which saves using the wrong type. I might make an extended list of exceptions on my fork and create a pull request for it.

HTV04, 2021-02-21:
I think I found the culprit. EEPROM v122 and EEPROM v124 appear to be set to the wrong values, they should be 64k and 8k respectively, instead of 8k and 64k. I made a commit on [my fork](https://github.com/HTV04/open_agb_firm/tree/kernel_experiments), would you be willing to compile it for me? I don't have the compilation setup for open_agb_firm ready yet.

If I'm right, all of the official games in the exception list can be removed, with the obvious exception of the Classic NES titles.

Also, why is EEPROM 4k referred to as 8k instead? I don't think there is an EEPROM 8k.

GBA save database used for reference: https://github.com/mamedev/mame/blob/master/hash/gba.xml

profi200, 2021-02-21:
The V<whatever> strings don't say anything about the actual save type. It's just SDK strings.

HTV04, 2021-02-21:
Are you sure? No-Intro and the MAME DB only contain references to EEPROM 4k, and the save type discussion post I linked before says there are only two types of EEPROM saves, 512B and 8KB.

profi200, 2021-02-21:
This is not emulation. The whole GBA including the cart (the ROM is stored in FCRAM) is actual hardware inside the 3DS SoC. The 3DS side just captures video frames, rotates them and outputs that to the LCD. There is no way to auto detect the save type at runtime. I also tried searching for common patterns in the code before but there are too many differences between SDKs. I want something solid without false positives.

HTV04, 2021-02-21:
I see. Anyways, I think it's safe to close this and move discussion to #9.

closed 2021-02-21 by HTV04