profi200/open_agb_firm #38: [QUESTION] Implementing game reset

Kyhze, 2021-05-03:
Is it possible to implement the soft game reset like on real hardware? 
The L + R + Start + Select reset combo.
It would be more convenient than having to restart the console each time! Not sure if the hardware allows it tho.

profi200, 2021-05-03:
If you mean switching games without rebooting it's not possible. I explained why in #21. A reset back to the BIOS is definitely possible.

Kyhze, 2021-05-03:
No I did not mean for switching games, just resetting the game currently being played. So yeah, a reset back into the BIOS. Preferably with a button combo for example.

foxsouns, 2021-05-03:
I'd love for there to be some sort of menu or something triggered via button combo, but I don't see it happening. Maybe X or Y could be a modifier key, with X+UP and X+DOWN for brightness, and X+START for reset to BIOS?

gyrorayo, 2022-10-04:
Actually, you can reset most games using A+B+Select+Start.