profi200/open_agb_firm #41: Banjo kazooie: grunty's revenge and grand theft auto advance can't save properly

bomba1749, 2021-06-28:


profi200, 2021-07-03:
It's a known problem that save type detection fails for many games. This is why #9 exists.

In the meantime if you are willing to compile yourself you can add the proper save types to [this override list](https://github.com/profi200/open_agb_firm/blob/kernel_experiments/source/arm11/open_agb_firm.c#L209).

bomba1749, 2021-07-05:
> It's a known problem that save type detection fails for many games. This is why #9 exists.
>
> In the meantime if you are willing to compile yourself you can add the proper save types to [this override [...]

Ok, would it be possible for the override list to be stored as a file separate from open_agb_firm? That way, people would be able to make games work, without needing to know how to compile.

profi200, 2021-07-16:
Possible? Yes. But not ideal. Would be very slow this way if you cram thousands of games into this list. For a few games it might be fine. Still would prefer using the save type db over this. Progress on the db is 1800 games currently out of over 2900.