profi200/open_agb_firm #43: [Bug] Agb Save format not compatible with GBArunner2

DraconicNEO, 2021-07-20:
One issue I have encountered when using this is that the Save Format isn't compatible with GBArunner2 and if you try to load Games with Saves that were played in GBArunner2 it wipes the save clean. The same thing happens when you load saves from games played on Open Agb firm.

Hopefully this issue can be fixed soon.

This issue Also doubles as a PSA. **Do not** try to load saves from games played in GBArunner2 in Open Agb firm. I recommend putting games you're going to play in AGB firm in a separate folder from games with GBArunner2.

profi200, 2021-07-20:
You are likely running into this issue described in the README:
> Note: EEPROM saves made by some emulators are incompatible because they have every 8 bytes block endian swapped.

I don't know how GBARunner2 saves EEPROM saves. Probably tries to be compatible with the majority of emulators. If this is the case there is a small tool to fix saves: https://gist.github.com/profi200/e06794d7561ed552c518b4b0b2f5f2f6

DraconicNEO, 2021-07-24:
> 
> 
> You are likely running into this issue described in the README:
> 
> > Note: EEPROM saves made by some emulators are incompatible because they have every 8 bytes block endian swapped.
> 
> I don't know how GBARunner2 saves EEPROM saves. Probably tries to be compatible with the majority of emulators. If this is the case there is a small tool to fix saves: https://gist.github.com/profi200/e06794d7561ed552c518b4b0b2f5f2f6
> It's not a bug. It's how the data is supposed to be laid out on EEPROM's. Unfortunately emulators got this wrong very early and never fixed it so the wrong save format has been used ever since.

You might want to talk to the maintainers of the GBAtemp wiki for GBARunner2, it states on [their page](https://wiki.gbatemp.net/wiki/GBARunner2#Compatibility_lists) that the save format is compatible with Open_AGB_Firm.
```
GBARunner2 is directly compatible with save files from the mGBA emulator, gpsp on 3ds and open_agb_firm.
```

profi200, 2021-07-25:
I don't have an account there and don't intend to make one. I heard the Everdrives do use the same layout as oaf does and they use .eep as file extension instead of .sav. If this is true i might switch to that to prevent any confusion.

DraconicNEO, 2021-07-25:
> 
> 
> I don't have an account there and don't intend to make one. I heard the Everdrives do use the same layout as oaf does and they use .eep as file extension instead of .sav. If this is true i might switch to that to prevent any confusion.

That is probably a good idea.

pdpapper, 2021-11-22:
Agreed. Using a different extension so that saves don't get overwritten seems important. This is also an issue with mGBA, as well as GBArunner2.

Better would be an option in the ini to choose the format. Personally, having compatible saves would more useful than them having the right endian-ness.

profi200, 2021-11-22:
Unfortunately what i said above turned out to not be true. The .eep saves from the Everdrive GBA use the same layout as commonly used .sav files.

So there are 2 options left: 1. Implement the wrong save layout or 2. use another custom extension.

DraconicNEO, 2021-11-23:
> Unfortunately what i said above turned out to not be true. The .eep saves from the Everdrive GBA use the same layout as commonly used .sav files.
> 
> So there are 2 options left: 1. Implement the wrong save layout or 2. use another custom extension.

Seems like option 1 is a better idea, sure it's less elegant but it's more widely compatible.

HTV04, 2021-12-08:
I agree, I think option 1 could be very convenient for transferring and using saves from GBA emulators.

If it's slow to convert between formats on the 3DS, it might be worth making an option that disables the save conversion for those who don't want it.

Masamune3210, 2022-03-15:
...what exactly does that have to do with the conversation?

DraconicNEO, 2022-03-16:
> ...what exactly does that have to do with the conversation?

Nothing, it's completely off topic.

YaketyJoe, 2022-09-21:
I would also like a more compatible save format, so I can easily transfer saves to my everdrive (to play on the TV)

branchus, 2023-08-25:
> Unfortunately what i said above turned out to not be true. The .eep saves from the Everdrive GBA use the same layout as commonly used .sav files.
> 
> 
> 
> So there are 2 options left: 1. Implement the wrong save layout or 2. use another custom extension.

I would like a more accurate save format(same as how hardware handles it)

Yes some early emulator uses wrong format but I can see nowadays some accurate gba emulator such as mgba or nanoboy advance's save are compatible with oaf's save

DraconicNEO, 2023-08-26:
> I would like a more accurate save format(same as how hardware handles it)
>
> Yes some early emulator uses wrong format but I can see nowadays some accurate gba emulator such as mgba or nanoboy [...]

I would suggest making a pitch to those emulators and playback hypervisors to support the correct format in that case.
The problem in this case is that not some, but most of them support the wrong format and this has already caused data loss for people, so it's more about supporting what most things support so people don't end up having a very bad day with savegame corruption than it is about hardware accuracy.

branchus, 2023-08-26:
> people don't end up having a very bad day with savegame corruption than it is about hardware accuracy.

I understand your point, 

I still think to have the accurate way to represent save is the right way. This needs to happen at some stage and by someone... The new emulators (mGBA, Nanoboy Advance and Skyemu, they all better than those old emulators, and probably more on the horizon) are now using the correct save format. The transition is already started, and we shouldn't turn back.

But agree with pdpapper, probably make some custom extension for save file, so the saves won't got overwritten. or to have an option to let user choose.

profi200, 2023-08-26:
I will probably end up using a different file extension for these saves. Using the same one just confuses everyone and you never know which format the save is in until you load the game which may delete the save.

HTV04, 2023-08-26:
> I will probably end up using a different file extension for these saves. Using the same one just confuses everyone and you never know which format the save is in until you load the game which may delete the save.

Personally I wouldn't do this just to alleviate the fact that emulators are using the wrong save format. It might actually make things more confusing by making it less obvious what the save file is, especially when most people don't care about the save type itself.

Instead, more emulators should move away from the wrong save type format as a default. I know mGBA and NanoBoyAdvance support the true format, but I'm not sure they do it by default, for compatibility with older saves.

branchus, 2023-08-26:
> I will probably end up using a different file extension for these saves. Using the same one just confuses everyone and you never know which format the save is in until you load the game which may delete the save.

Be lazy for a couple of years while the new emulators start catching up. This won't be an issue anymore. Lol

DraconicNEO, 2023-08-27:
> Be lazy for a couple of years while the new emulators start catching up. This won't be an issue anymore. Lol

Will the GBArunner2 compatibility layer catch up though? That's one of the bigger problems here because people might use them both on the same device or same SD card in different Devices (i.e. DSi).

lucasmr, 2023-10-11:
Just tried reproducing this, but I was able to use a save file from open_agb_firm into GbaRunner2 and vice-versa. Tested with two different games, so is this still an issue? I just had to copy the .sav file to the correct folder.