profi200/open_agb_firm #44: [ENHANCEMENT] GBA FRAME BORDERS

Zarke669, 2021-09-19:
is it possible to add frame borders? the same way it works on the game boy color virtual console.

Zarke669, 2021-09-19:
[image]
[image]
[image]

profi200, 2021-09-20:
It is possible but there are a few reasons i didn't attempt this yet:

* It takes precious VRAM space.
* It requires drawing another huge quad with the GPU which may or may not destroy the low latency i'm aiming for (needs testing).
* It requires new GPU command lists. I can't yet change what the GPU renders dynamically.

I will keep this open until i get to testing it.

Zarke669, 2021-09-20:
is it possible to leave a screen centered as seen in the gba virtual console?
[image]

Zarke669, 2021-09-20:
example I did with the retroarch.
[image]

Zarke669, 2021-09-20:
[image]

profi200, 2021-09-21:
Position can be arbitrary but again needs changes to GPU rendering.

profi200, 2023-08-10:
It's now implemented and for now it's raw format only.
Place "border.bgr" in "/3ds/open_agb_firm". It must be a 400x240 border with raw RGB format (color channel order BGR). ImageMagick can convert this very easily like so:

`magick ./border.png -rotate +90 ./border.bgr`

The frame is displayed exactly centered meaning 80 pixels left/right and 40 top/down are free for the border. As reference i uploaded this literally shit border:
https://i.ibb.co/Bnh6BxF/border-poop-rotated.png

GorGylka, 2023-08-20:
please build a new release with custom border support!

andromalandro, 2024-01-09:
can someone upload the .bgr file i cant convert the image :(

RandomMan25XX, 2024-01-13:
> can someone upload the .bgr file i cant convert the image :(

If you're on something that uses ImageMagick 6 you'll want to substitute the "magick" command with "convert", if that's not your exact issue however I'd be happy to help you sort that out

BrazilKing, 2024-01-22:
Hello, I have been trying to convert my own borders but I can't get Magick to convert the my .png files. I tried with a website that converts PNG to BGR but open_agb_firm does not show the border. I would like to receive more help how to convert the image properly, thank you.

BrazilKing, 2024-01-22:
Got it working through another way but still I didn't get it to work with Magick. It doesn't pick up the command line no matter what I try. The render issue was caused by launhing open agb from the 3ds main menu instead of luma.

BrazilKing, 2024-01-22:
update 2: 

websites for converting PNG to BGR do not provide a correct render either. Is there any other way for me to convert from PNG to BGR on windows? I like to make multiple designs.

profi200, 2024-01-23:
You need to rotate the border. The frame buffer on 3DS is sideways because the LCD is sideways. That's what the "-rotate +90" parameter does.

And a little side note but looks like older versions of Imagemagick don't have the "magick" command. Instead use the "convert" command.

BrazilKing, 2024-01-23:
> You need to rotate the border. The frame buffer on 3DS is sideways because the LCD is sideways. That's what the "-rotate +90" parameter does.
> 
> And a little side note but looks like older versions of Imagemagick don't have the "magick" command. Instead use the "convert" command.

Thank you, it resolved the matter! I rotated the image manually and then converted it to .bgr using a .png to .bgr converter website and now it is being rendered properly on my 3DS.

BrazilKing, 2024-01-23:
[image]
[image]

rhymin36, 2024-02-05:
I made some borders today so thought I'd share.  All 6 official colours :)

[download](https://file.io/wyoGxX0qh7Mg)

thanks for open_agb_firm love it

BrazilKing, 2024-02-06:
I wonder if Rhymin36 made the screen border himself or he just copied mine... Anyways here are some awesome overlays that I did make myself.

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

My private fb group for 3DS custom (theme) designing and requests -> https://www.facebook.com/groups/549132360461223/?ref=share
3DS Custom Themes Wizard

BrazilKing, 2024-02-06:
[image]

rhymin36, 2024-02-06:
Not sure what you mean by that comment bruv, mine doesn't look anything like the ones you've posted.  I made them in photoshop yesterday...

For everyone else, I made another version today without buttons, think it looks cleaner, If anyone would like them just shout.

Cheers

BrazilKing, 2024-02-06:
Awesome, keep up the great work 🤘🏽

rhymin36, 2024-02-06:
You too ;-)

AmazingJorgito, 2024-02-25:
> I put the .bgr file in the open_agb_firm folder in the 3ds directory and it doesnt boot the games with the border or in 1:1. Can anyone help?

On your sd card > 3ds folder > open agb firm folder you have to edit the config.ini. Change scaler from 2 to 0, save and done.

Jvisawesome, 2024-02-27:
> > I put the .bgr file in the open_agb_firm folder in the 3ds directory and it doesnt boot the games with the border or in 1:1. Can anyone help?
> 
> On your sd card > 3ds folder > open agb firm folder you have to edit the config.ini. Change scaler from 2 to 0, save and done.

Thank you so much! Works like a charm and the colors look super vibrant!

Jvisawesome, 2024-02-27:
> I wonder if Rhymin36 made the screen border himself or he just copied mine... Anyways here are some awesome overlays that I did make myself.
> 
> ![GBA_overlay_basis_preview](https://private-user-images.githubusercontent.com/157328178/302615057-7be1b1c0-4110-4e06-bfc6-ec1fcf971c52.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDkwNTQ0NzQsIm5iZiI6MTcwOTA1NDE3NCwicGF0aCI6Ii8xNTczMjgxNzgvMzAyNjE1MDU3LTdiZTFiMWMwLTQxMTAtNGUwNi1iZmM2LWVjMWZjZjk3MWM1Mi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMjI3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDIyN1QxNzE2MTRaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1iM2Q2NjM1MDg2YmEyNTBmZmMxZDVjZDYzYTBhNDhlMDQyNDM0OGIwMzM3ODYxNTA2MDE4NTNmMjNkMzY5MzQzJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.StmUzAzZimCmny3SULaeYPqpfQ2ag1HakPqATozbu_E) ![GBA overlay mario preview](https://private-user-images.githubusercontent.com/157328178/302615060-19b83e64-21fd-46d7-bebf-09fba75bfcf8.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDkwNTQ0NzQsIm5iZiI6MTcwOTA1NDE3NCwicGF0aCI6Ii8xNTczMjgxNzgvMzAyNjE1MDYwLTE5YjgzZTY0LTIxZmQtNDZkNy1iZWJmLTA5ZmJhNzViZmNmOC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMjI3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDIyN1QxNzE2MTRaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT05OGNjNDU3YmRhOGE1Y2Q4ODVjODczZjE5ZTZmMDJiODMzZGViZGY0ZDE3NzBiZTEyZTgwZDlkNjEzZDc2M2U2JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.wDvK38JunjUQlnI-ppks0IKEN_r8VavjudAJmmtrEiI) ![GBA overlay zelda preview](https://private-user-images.githubusercontent.com/157328178/302615062-324acc8a-4b8e-4202-9676-419a9c6269b9.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDkwNTQ0NzQsIm5iZiI6MTcwOTA1NDE3NCwicGF0aCI6Ii8xNTczMjgxNzgvMzAyNjE1MDYyLTMyNGFjYzhhLTRiOGUtNDIwMi05Njc2LTQxOWE5YzYyNjliOS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMjI3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDIyN1QxNzE2MTRaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0zMjA2YzU2ZGQwNTMyOGFjYmM4OTY0Yzc4NTZkYWVjMzE4YWViY2I2NGU1NTYwZWUyMzM4MWNjOTZkYTY4MTJiJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.tlrIvcSVGRFeF3ZS1cG7cjxyDnSsOV1AJzvUlTDBLZo) ![IMG_20240131_233920_033](https://private-user-images.githubusercontent.com/157328178/302615065-fb15f2b1-6f80-499a-adae-c49aac82d555.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDkwNTQ0NzQsIm5iZiI6MTcwOTA1NDE3NCwicGF0aCI6Ii8xNTczMjgxNzgvMzAyNjE1MDY1LWZiMTVmMmIxLTZmODAtNDk5YS1hZGFlLWM0OWFhYzgyZDU1NS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMjI3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDIyN1QxNzE2MTRaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1iNmU1MGQ4OWJhMDkyNGZmZGQ4YTI5ZWQ3MzUwZjMyYTVjZTZmODYzZGZmN2NiODcxYWJiODcwMjlmODc5NjJmJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.Ca-dx6wk8G8e0kuIkAzUlBae_ndp6JpskYF8l_vJcrM) ![GBA_overlay_pikachu_preview](https://private-user-images.githubusercontent.com/157328178/302615071-e9b460ba-7113-4ef5-8249-507b8abea363.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDkwNTQ0NzQsIm5iZiI6MTcwOTA1NDE3NCwicGF0aCI6Ii8xNTczMjgxNzgvMzAyNjE1MDcxLWU5YjQ2MGJhLTcxMTMtNGVmNS04MjQ5LTUwN2I4YWJlYTM2My5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMjI3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDIyN1QxNzE2MTRaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0xN2M1MDNmMTRlMjIxM2U4Y2JmM2QyMTZjNzg1NDZjNjBhZWIxYzY1ZjUwMmQxZDhiNWFlNDEwNDljNDBhYmQzJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.6RG9CYqBcYgNmDp9pUthpJ1YzG0Qg7NYlRKLDFf7Rrw) ![IMG_20240201_183952_321](https://private-user-images.githubusercontent.com/157328178/302615073-12d383ad-30a7-4ea2-8ddb-646893238a1f.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDkwNTQ0NzQsIm5iZiI6MTcwOTA1NDE3NCwicGF0aCI6Ii8xNTczMjgxNzgvMzAyNjE1MDczLTEyZDM4M2FkLTMwYTctNGVhMi04ZGRiLTY0Njg5MzIzOGExZi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMjI3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDIyN1QxNzE2MTRaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT02YWJlNWUxZmFhZmFlNWNhN2YwNGY5NWJkM2EzY2YwYzE4YzFmNTNhMjk1OGU0MTdkNWVjYzFkNTVjMzE3MDM2JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.RHLU01yt8gy7SijDAxIqmqCMrLeAn-NyDsFhYyTEkaU) ![IMG_20240201_183954_522](https://private-user-images.githubusercontent.com/157328178/302615079-f6186d28-2815-46be-aebe-67eb14b8ec6f.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvb [... cut ...] aJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT05N2RiYjVmYWY3M2E0M2MxYzlkZDIxYmRlYjQxYzllOGM4NzJjYmJhNzZlYmFiMmJhN2M2ZjM3Zjg1ZmRhMGZkJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.ErNdR6NvZYp31xtUDfIogMOcickFVcwUIMzwklmTNo8)
> 
> My private fb group for 3DS custom (theme) designing and requests -> https://www.facebook.com/groups/549132360461223/?ref=share 3DS Custom Themes Wizard

Do you mind sending download links for these in the .bgr format? They all look amazing

Farhan1379, 2024-07-07:
How did you do that?
I want to know too

Mario-Valiante, 2024-11-30:
A gba SP frame I made:

[image]

Rename gba_sp.bgr to border.bgr

[gba_sp.bgr](https://drive.google.com/file/d/18GibsiYp7Y9y6QQeQMaWGukCvrnysTTV/view?usp=sharing)

MMX-64, 2025-02-28:
Since people seem to be posting borders they've made here I guess I will to.  It's a recreation of the border used on the Wideboy 64 AGB.  It's not pixel perfect because I had trouble finding a direct rip of the N64 version, but it shouldn't be noticeable.

[image]

GitHub won't let me upload .bgr files directly so its packed as a zip file

[border.bgr.zip](https://github.com/user-attachments/files/19020377/border.bgr.zip)

Scokera, 2025-03-15:
Does it work on old 3ds?

akshatshah256, 2025-04-30:
thanks to @BrazilKing , i traced one of the overlays you posted and went from there. used a lot of assets from the super gameboy borders and some Gameboy pokemon in general

[image]

Iruel13, 2025-06-15:
These borders are a really great feature, thanks for that. I also made some for myself. Nothing special but someone might like them too:

[image]
[image]
[image]
[image]

[Iruel_open_agb_firm_borders.zip](https://github.com/user-attachments/files/20744873/Iruel_open_agb_firm_borders.zip)

JCM-devv2, 2025-11-11:
Can someone make the gameboy player borders a thing too

Mario-Valiante, 2025-11-13:
> > Can someone make the gameboy player borders a thing too
> 
> Well, there are several Problems with that. For once, the GBP Borders have a resolution of 608x448 (19:14) while the 3DS Top Screen is at 400×240 (5:3). So the size is to big and the ratio is off too. Therefore it is impossible to transfer the GBP borders 1:1 and lossless to open_agb_firm. The other problem is that afaik there is no way of extraction the textures from the boot disk, so you have to rely on HD screenshots taken with a capture device, if you want to have top sorce material. A little unknown is the fact that Pokemon Box for GameCube also have GBA borders in the same way as the GBP.
> 
> And that is how it would look like if you put one of those borders on a 3DS screen:
> 
> <img alt="Image" width="400" height="240" src="https://private-user-images.githubusercontent.com/62846064/513601992-8f48754e-5708-4d2a-847a-9d835d6ff659.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NjMwNDQ2MTUsIm5iZiI6MTc2MzA0NDMxNSwicGF0aCI6Ii82Mjg0NjA2NC81MTM2MDE5OTItOGY0ODc1NGUtNTcwOC00ZDJhLTg0N2EtOWQ4MzVkNmZmNjU5LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTExMTMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUxMTEzVDE0MzE1NVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTMzZGE3NjllMGY1ZmNmZGExZjMwYzBmMWE2OGViZTVhMTFiMDNhMTdkMzM4YTgxOTNkMmVlYjZiYzIxZjdhNzgmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.5sk0nlLcLTSyEM_Fqnr-trqxpg34tlxC55dO-t0ZQjQ">
> So, unless you design the now blank space yourself or use AI to fill in the blank space, I think it doesn't worth the hassle.

Having blank space on the left/right is not a dealbreaker in my opinion (see my gba sp frame a few posts above), and your example seems quite nice to me.
Beside this, I was thinking how cool it would be to be able to associate each game with its own specific frame.

Iruel13, 2025-11-14:
> Can someone make the gameboy player borders a thing too

> Having blank space on the left/right is not a dealbreaker in my opinion (see my gba sp frame a few posts above), and your example seems quite nice to me. Beside this, I was thinking how cool it would be to be able to associate each game with its own specific frame.

Well here you go:

[image]

[image]

[image]

[image]

[Iruel_open_agb_firm_borders_gbp.zip](https://github.com/user-attachments/files/23556790/Iruel_open_agb_firm_borders_gbp.zip)
The Archive includes all 20 borders from the GBP and both from PKM-Box. Plus several custom one. please read the Note.txt inside the zip.

The other day I found a website that provides the lossless assets from the GBP borders, so I spent a couple of hours to put them together. I'm not a designer so that was more like putting together a puzzle. I hope you guys like the end result.

kek170, 2025-11-30:
> Since people seem to be posting borders they've made here I guess I will to. It's a recreation of the border used on the Wideboy 64 AGB. It's not pixel perfect because I had trouble finding a direct [...]

I love this. have you found it with a better resolution? or at least a simple and clean border similar to this one

luvz888, 2025-12-04:
> Since people seem to be posting borders they've made here I guess I will to. It's a recreation of the border used on the Wideboy 64 AGB. It's not pixel perfect because I had trouble finding a direct [...]

I really love this. Have you found it in better resolution? Or at least a border minimal style similar to this one?

ezrawalker745, 2026-01-11:
[I made my own Windows Style Border!](https://github.com/user-attachments/files/24553130/Pink.Windows.Border.zip)

[image]

Though i do have my own border related question. is it possible to add custom border support alongside game specific ini files?

LEsqee, 2026-02-28:
> [image] [image]
> 
> (Outdated designs)

Can you share the second border?