profi200/open_agb_firm #45: [QUESTION] Can a loaded (custom) .gba access X, Y buttons?

draincode, 2021-09-21:
Can a  GBA rom loaded with open_agb_firm access the extra buttons (X, Y, perhaps even touchscreen?)?

I've tried reading the halfword at 0x04000136 (NDS EXTKEYIN) with no luck. I've also tried to read it from the addresses REG_HID_PAD, REG_LGY_PAD_VAL (which maybe makes no sense from the .gba side).

One thing I noticed is that pressing Y freezes screen for some frames. In every game, even at boot logo. Why is that?

profi200, 2021-09-22:
No, at least not via standard input. You can only access [these buttons](https://problemkaputt.de/gbatek-gba-keypad-input.htm) on the GBA side.

Y is the screenshot feature and it's unintentionally causing the freezes. This has been changed to Y + SELECT.

draincode, 2021-09-22:
Ah, I see. Thanks for the info!!

closed 2021-09-22 by draincode