profi200/open_agb_firm #48: White screen for games with no save type

thomas1013, 2021-11-27:
With the last release, when I launch a game with no save type (ie: Finding Nemo, Britney's Dance Beat...), it leads to a white screen and I can't play. I have to either disable gba_db.bin or mess with bool saveOverride true and then select another random save type which is not 15.

profi200, 2021-11-30:
I will look into it later. Which no-intro release numbers are these?

thomas1013, 2021-12-02:
Actually this issue happens with every game which has the save type defined as 'None (15)' in the gba_db.bin file. When I select another save type with bool saveOverride true it works.

I also tried to select the 'None (15)' save type on other games which have a save type to see if it would crash too, and it did for most games. The no-intro release numbers for Finding Nemo & Britney's Dance Beat are respectively 2664 & 0460.

profi200, 2021-12-05:
The entry for 0460 is identical between my db and the one from the current release. So it's either buggy binary search (doubt it) or the entries in the db are not ordered correctly.

@HTV04 ^

HTV04, 2021-12-05:
If it was an issue with the way the DB entries are ordered, then the saveOverride method would have worked, but thomas1013 noted that setting the save type to "None" causes the games they tested to hang.

@thomas1013: Which games have you tested that hang at a white screen?

mikedee397, 2021-12-06:
I noticed some games boot to a white screen. Depending on name and possibly location of the file, it will either boot correctly or go to a white screen.

Example of game that goes to white screen:
/gba/1 USA - P-S/Sega Arcade Gallery (USA).gba

Same file boots correctly from here:
/roms/gba/1000-2000/1877 - Sega Arcade Gallery (U).gba

The files are identical with binary compare.  Not sure if there is something else in the folder that could trigger this issue.

HTV04, 2021-12-07:
What happens if you remove the spaces in the first folder name? Looks like this might be causing an issue.

mikedee397, 2021-12-07:
Removing the spaces doesn't seem to help, still goes to white screen. I also tried just moving the file to the root folder and it also went to a white screen.

HTV04, 2021-12-07:
If you move the file stored in the other folder to the root of your SD, does it also hang at a white screen?

mikedee397, 2021-12-07:
I just realized what's going on here. It boots because there is a 64kb .sav file in the working folder. It must have been created by an older build of open_agb_firm. When that file is present, the game works. When the file isn't there, the game goes to a white screen.

HTV04, 2021-12-08:
Interesting. And it works when the save type is set to none?

profi200, 2021-12-08:
That doesn't sound right. I smell SD card fs corruption. Or a weird bug with the database. When the save type is truly set to none it doesn't matter if a save file exists.

mikedee397, 2021-12-08:
I have tried 3 SD cards now and get the same result.

thomas1013, 2021-12-08:
> If it was an issue with the way the DB entries are ordered, then the saveOverride method would have worked, but thomas1013 noted that setting the save type to "None" causes the games they tested to hang.
> 
> @thomas1013: Are there any other games you tested that hang at a white screen? The games I tested that don't have a save type work fine for me, so these games may be looking for an SRAM even though they don't need it.

To answer to your question, yes, I tried with several games with no save type : Finding Nemo, Britney's Dance Beat, Monsters Inc., Barbie and the Magic of Pegasus and The Incredibles (yes I love shitty license games haha). They all hung to a white screen. 

For games with a save type + bool saveOverride true + setting the save type as 'None' for testing purposes, these are the games which all hung to a white screen: Wario Land 4, Super Mario Advance 2, Shrek 2, The Sims Bustin Out, Pokémon Emerald & Mother 3. F-Zero GP Legend, Kurukuru Kururin, Kururin Paradise, Lizzie McGuire 3 & Pokémon Pinball worked without issues. Super Mario Advance 4 (Wii-U version) launched but displayed a 'Save corrupted' error.

profi200, 2021-12-08:
Comparing to games that do require save hardware to work correctly isn't very useful when setting their save type to none. That said i'm going to run a few tests and see what is happening.

profi200, 2021-12-08:
Test ROM:
"de0e0152d55666b29cd2d414761852b7ae7645189a0db7c5e920ecb331c9eb82  1853 - SEGA Arcade Gallery (USA).gba"

2 GB SD card on an o3DS.

Debug build:
With the 2 file paths given in https://github.com/profi200/open_agb_firm/issues/48#issuecomment-987214653 both copies of the game boot fine. The initial copyright screen takes a bit until it vanishes.

Release build:
Both hang on white screen.

I'm not sure what's the cause but i think it's link time optimization (LTO) related. Not sure why or how it causes issues here.

commit 434f714e (2021-12-09): Fixed white screen issues with save type none games. Fixes #48.
closed 2021-12-09 by profi200