profi200/open_agb_firm #49: add gba game that can switch between titles with home button

Alpha-404, 2021-11-29:
basically gba game to load other gba games so you dont have to reboot

Masamune3210, 2021-11-29:
likely impossible, considering i believe the home button is mapped to closing the entire thing

profi200, 2021-11-30:
The way you describe it is not possible. The ROM is stored in FCRAM (external DRAM) and all access to that RAM is blocked once GBA mode has been entered (GBA hardware gets exclusive read access to it). Meaning i can't overwrite the ROM in any way once in GBA mode.

I however plan on allowing reboot to oaf with the help of the installed bootloader.

Rose22, 2022-11-06:
can you at least create a button shortcut that you can press during a game that will reboot the 3ds straight back into open agb firm so you can pick another rom to play?