profi200/open_agb_firm #51: [Request]Can you add support for more languanes?

butenewbee, 2021-12-12:
When i using it,the ROM Had to be named in English,it not very convenient fou us.Can authors add support for other languages to display other languages in ROM list,suca as japanese,chinese,korean,etc.
plz

profi200, 2021-12-12:
I don't understand what you are trying to say. If you want a different language use the ROM of the game in your language.

piratesephiroth, 2021-12-12:
> I don't understand what you are trying to say. If you want a different language use the ROM of the game in your language.

I think that person is talking about supporting non-english characters in the file names

profi200, 2021-12-12:
If this is the case then not before i added a proper GUI. The console just can't handle anything but ASCII.

butenewbee, 2021-12-13:
> We really hope to add this feature, it is really helpful for non-English speaking players, we hope you can consider it, thank you again for your reply.