profi200/open_agb_firm #60: PocketNES doesn't save.

yokuma123, 2022-03-31:
PocketNES doesn't save.

It would be so nice if you could take time to  fix the issue.
I want to use load and save states.

https://www.dwedit.org/gba/pocketnes.php

yokuma123, 2022-04-01:
I checked /3ds/open_agb_firm/config.ini and saw 14: SRAM 256k is set in Advanced.
So I think pocketnes save type is already set to SRAM.
But after rebooting 3DS, all saved states are erased.
What else should I try? Should I create romName.ini and add something?

profi200, 2022-04-03:
I don't know what's the issue to be honest. I can only recommend using the ROM manager that comes with this emulator and avoid packing too many games into 1 ROM because, as mentioned, there is only 32 KiB of SRAM instead of 64 or 128.

yokuma123, 2022-04-05:
So I tried Super Mario Bros. 3 and PocketNES.gba stored saved states. 

I also tried Super Mario Bros. 1 and PocketNES.gba could not store saved states of Super Mario Bros. 1.
All saved states were lost after 3DS rebooted.

Perhaps Super Mario Bros. 1 did not have SRAM so that is why pocketNES loses the saved states.
I think Super Mario Bros. 3 had SRAM.

If I do the same tests on Boycott Advance, the results are same. Super Mario Bros. 1 saved states were lost. But Super Mario Bros. 3 saved states remained.
If I do the same tests on EZ-Flash IV, EZ-Flash IV never lose saved states of Super Mario Bros. 1. and Super Mario Bros. 3