profi200/open_agb_firm #61: Video scaler options don't appear to do anything

RueGorE, 2022-04-18:
I tried changing the video scaler options in the config.ini between 0, 1, and 2 but the output appears exactly the same in every case. I even dumped the screen output on each mode and as far as I can tell they're not different from each other in any way. I would prefer to play with a bilinear filtered output. Is this config option just not supported yet?

profi200, 2022-04-18:
Make sure every config option is in its own line. The last release creates a default config file where one newline is missing.

RueGorE, 2022-04-18:
Yes, every config option is on its own line. Despite this, changing `scaler` still doesn't do anything. BTW I am using the latest release. Am I correct in assuming scaler option 0 is supposed to output a 1:1 image, 240x160 pixels? It just remains at the full size of the 3DS screen.

profi200, 2022-04-18:
See the comment in [this line](https://github.com/profi200/open_agb_firm/blob/master/source/arm11/open_agb_firm.c#L69) or the readme for the scaling options.

Make sure the config is in the right place. It's "/3ds/open_agb_firm/config.ini" for a while now. If this isn't the problem then i don't know what's wrong. Maybe reformat your SD card.

RueGorE, 2022-04-19:
I downloaded a fresh new copy of oaf onto another 3DS. Ran oaf for the first time on it to have it automatically create `config.ini` in the default `sdmc:/3ds/open_agb_firm/` directory. As it had with my first 3DS, the config file did not have the `scaler` option listed anywhere, so I dropped a newline and put it between `[video]` and `gbaGamma=2.2`. I entered `scaler=1`. Saved and ran oaf. And the output still isn't bilinear.
Then I changed it to `scaler=0` and also changed from `directBoot=false` to `directBoot=true`, then ran oaf again. It skipped past the BIOS boot logo but the video output is still the same.
It's like oaf is ignoring the `scaler` parameter completely. Other options like `backlight` work as expected.

profi200, 2022-04-21:
It seems like you are using an outdated version. The default config should contain the scaler option. Download the latest version again from [here](https://github.com/profi200/open_agb_firm/releases) and try again. Keep the newline bug in mind i mentioned earlier.

ChainSwordCS, 2022-04-24:
> It seems like you are using an outdated version. The default config should contain the scaler option. Download the latest version again from [here](https://github.com/profi200/open_agb_firm/releases) and try again. Keep the newline bug in mind i mentioned earlier.

The commit to add configurable scaling was done in January 2022, but the latest Stable Release is older (December 2021 IIRC).

xgwmir, 2022-04-24:
I tried changing the video options in the between 0, 1, and 2 but the output appears exactly the same in every case

profi200, 2022-04-24:
Ah, yeah. I totally forgot the latest release doesn't contain this improvement. I will see if i can make a release later.

profi200, 2022-04-25:
A new release is up with the missing scaler option.

closed 2022-04-25 by profi200

ethanchen, 2022-05-15:
The scaler is till unavailable on my old 3DS with [this file](https://github.com/profi200/open_agb_firm/releases/download/alpha_2022-4-25/open_agb_firm_alpha_20220425.7z) .  I also try to delete the old `config.init` and reboot to check the default `config.init`, there's no `scaler` option in the generated file.

ethanchen, 2022-05-22:
Sorry, my mistake. I was put the .firm in the wrong path. It works  right now.