profi200/open_agb_firm #66: RTC becomes unsynced

WholesomeIsland, 2022-05-17:
The rtc clock becomes out of sync while playing. If i open up openagbfirm and play for 30 minutes, even with the clock set right before playing, it will be 30 mins out of sync. Judt supplying power to the rtc so it keeps count might fix it

profi200, 2022-05-18:
Not enough infos. Which game and did you set the correct date and time using GodMode9?

WholesomeIsland, 2022-05-18:
I set the date and time with GodMode9, then played pokemon firered and mario & luigi superstar saga. The issue happened with both games.

profi200, 2022-05-19:
Both of the games you mentioned don't have the RTC feature at all.

profi200, 2022-05-19:
Which RTC are we talking about? The RTC of the 3DS? I can't reproduce this at all. The 3DS RTC stays in sync here.

I would recommend the following steps if the RTC in the MCU is the issue:
* Remove battery and hold power for 30 seconds. Then wait 5 minutes.
* Insert the battery back and open GodMode9 on first boot to correct date and time.
* Boot into HOME menu, open settings and fix the date and time there.

If the time still desyncs the only explanation i have is hardware fault.

WholesomeIsland, 2022-05-19:
I will try that, thanks for the help!

closed 2022-05-28 by WholesomeIsland