profi200/open_agb_firm #73: For DS

CrezyDud, 2022-06-10:
Using GbaRunner is not so good and makes me sad
The Compatabillity is Garbage,
Per Game Hacks are needed
and it's slow (when it's not, sound is delayed by 1.5s)

I found someone asking if it's possible for the DS-i, you said it wasn't
But the DS(not -i) can run gba

If it were possible to port this to the ds, there would finally be a great way to run gba without flashcard (those were $$)

profi200, 2022-06-10:
The DS (lite) does have the GBA hardware but it does not have any hardware cartridge emulator like the 3DS does so this is not possible. Even if it had one it would be a different project.

closed 2022-06-10 by profi200

kevincrans, 2022-06-17:
I know it's closed, but does the DSi have such cartridge emulator?

Dartz150, 2022-06-17:
> 

GBARunner is a POC that was paused due to the developer working on other projects and school, but is open source, so you are welcome to improve it.
If you mean SRAM patching, that was deprecated many versions ago, as it is now done automatically. 
The sound delays are only happening on DS Lite, on DSi, it uses the DSP to emulate sound, so the delays are not present.

So if you want GBA games running on the DS Lite, the best option is to get a GBA flashcart.