profi200/open_agb_firm #77: can't compile firm

cooolgamer, 2022-07-16:
Hi, I tried compiling openagbfirm because my brother broke his top screen but that's a other thing.
So I want to edit it to make it work on his console but I can't compile it even wihout editing anything, first, it says that firm builder is "not a file or directory", well, I use firmtool so idk why it didn't detected it, and, after editing makefile to make it use fiitool, it says:
"firmtool build: error: argument -C/--section-copy-methods: invalid choice: '0' (choose from 'NDMA', 'XDMA', 'memcpy')"
Is it possible to fix this? Thanks

profi200, 2022-07-17:
There is currently no way to make it output video on bottom screen unless you want to dive into the hell of low level GPU programming.

As for compiling it expects ctr_firm_builder to be reachable via the $PATH environment variable. Build ctr_firm_builder and then do "export PATH=/path/to/ctr_firm_builder:$PATH" where you have to replace "/path/to/ctr_firm_builder" if that was not obvious.

cooolgamer, 2022-07-19:
Thanks but the thing is I use firmtool and not that firm builder, I installed it using the python scrii from it and it works fine for compiling other firms but not this one...
And if you are saying that's not possible, I guess that's useless I try to compile it then... And just use mgba because that's the only thing I can do

cooolgamer, 2022-09-14:
fixed by typing "make USE_FIRMTOOL=1" on cmd

closed 2022-09-14 by cooolgamer