profi200/open_agb_firm #78: [Feature Request] Add option to use rom folder for save files

syndenbock, 2022-07-17:
I have configured my emulators to write save files in the same folders as the rom files to be able to quickly copy roms + saves to another system. Therefor it would be nice to have the option in OAF too.
Due to other emulators writing the wrong EEPROM format it might be useful to only enable overriding same-folder save files if they could be read correctly to avoid overriding existing saves with fresh ones.

profi200, 2022-07-17:
The reason why i separated ROMs and saves is because i think this is easier to organize. If one folder is full of mixed files it's annoying to find and copy out the save files for backups.

It's not feasible to try and detect wrong EEPROM saves. It doesn't work. What i can do is use a different file name extension so only emulators understanding this format can load these saves.

syndenbock, 2022-07-17:
I think both methods have their benefits depending on how you handle them - hence why I think the option would be a nice thing to have (default can obviously still be to have them in a separate folder).

I really like the idea of having a different file extension for fixed EEPROM saves as it makes it easy to recognize which format they have and creates more possibilities of dealing with save files.
I have no idea about the technical insides of loading save files but with having a different file extension would it be possible to convert .sav files to the correct format in case no save with the different extension exists?
So basically when reading save files the logic would be:
1. Does a save file with the new extension exist? > Load that save file
2. If not, does a .sav save file exist? >Convert said save file to the correct format with the different file extension
3. If neither exist create a new save file

That would obviously create the need to rename all save files created by previous versions to be renamed to the new file extension to prevent them from being converted.

gricard, 2023-03-26:
@syndenbock I feel the same way. I wanted to be able to load my rom/save combos in this the same as I can using twilight menu++ and not have to move files around. I was able to patch the source to make it load the .sav from the same directory as the rom. Here's a [gist](https://gist.github.com/gricard/5b175ce738a9878b3d6d2f5be269c02c) with my changes. They're extremely basic. If I have some time at some point I could maybe create a proper PR with a config option.

PR #209 (2024-10-28): Add an option to use current rom directory for config and saves
Closes #78
closed 2024-10-28 by profi200