profi200/open_agb_firm #79: [request]Get save type from rom

a92126, 2022-07-28:
In the latest firm, the program will use binary-search from file `gba_db.bin` by comparing the SHA1 to match the SAVE TYPE of game.

It works well for the public roms, however there are lots of Chinese localisation from fan in China, and all these roms will meet the 
saving problems for different SHA1.

In fact, the gba_db.bin includes the serial of gba file, and the serial field is unique with gba roms, too. Why not search the save type by comparing the serial field? I think it could help to solve this problem.

note: the serial is store at 0x000000AC in the gba file.

profi200, 2022-07-29:
Binary search is used for speeding up the db search and not having to load the whole db in RAM. The db is mixing both retail and prototype/debug ROMs to support them all. The debug/prototype ROMs often have the same game code/serial as the retail one but they use SRAM saving which means this would not work.

I actually wanted to do a fallback to linear search for the game code/serial but this requires significant changes to the db.

Henny022, 2023-02-17:
for most games the save type can be determined by a string in the rom, ie. zelda minish cap (using 64k EEPROM) has a `EEPROM_V124`.
This would support roms that have no entry in the DB or romhacks that change the save type, but not the game code.
(The last one is whats happening with the minish cap randomizer)
A downside is, this method does not detect save size (at least not for all save types as far as I'm aware).
And would require scanning the whole rom for the string.
But it could make a nice fallback option for entries missing in the db

profi200, 2023-02-17:
oaf already uses that method searching for strings as fallback but this is very unreliable. EEPROM 8k and 64k are not only different size but there are differences in the way the game talks to the EEPROM chip as well.