profi200/open_agb_firm #81: Displaying position when scaler=0

a92126, 2022-08-14:
When I set the scaler to zero, the image with original size is displaying in the bottom of the top screen.

Is there a way to make the image displaying in the center of the top screen? Or some source code to generate
the gbaGPUInitList?

profi200, 2022-08-14:
The code used to generate the GPU command list is [here](https://gist.github.com/profi200/e31df7d31b8c2cdccc0608aaa71681fc). However open_agb_firm does patch the cmd list at runtime and multiple addresses in the cmd list need to be fixed.

Since i have no proper GPU driver yet adding more patches for centering would be a bit much.

a92126, 2022-09-10:
> The code used to generate the GPU command list is [here](https://gist.github.com/profi200/e31df7d31b8c2cdccc0608aaa71681fc). However open_agb_firm does patch the cmd list at runtime and multiple [...]

From the source, the scaler 0 and 1 implemented by command list, and 2 by legacy frame buffer translate. The output will get
some burrs on edge, which cause by the lack of sampling. I wonder if a pixel shader is possible to filter the source of gba video. I think a scaler of 3 can apply a xBRZ shader to create a 720x480 render target and downscale it to 360x240.

profi200, 2022-09-10:
Unfortunately no fancy shaders are possible. There are only fixed fragment shader stages (6 tex envs). No real fragment shaders. I wanted to do a GBA color shader similar to what mGBA offers but it's very tricky to do without fragment shaders.

profi200, 2022-09-11:
No reason to close this. The original request of centering the output is valid and doable.

a92126, 2022-09-12:
Oh, I forgot the original request. Thanks.

homosexuayla, 2023-08-31:
this is fixed now, isn't it?

profi200, 2023-08-31:
I guess? But it's still hardcoded.