profi200/open_agb_firm #82: Real-Time Clock doesn't work right

FIFFY-Fede, 2022-08-30:
In pokemon emerald there's an npc who tells you how many weeks have passed since the start of the game (real life weeks)
is said to me that 1182 weeks passed... more than 24 years, thats not possible hahhaha

for context i use the latest stable realase, the sav file were created with open agb firm (latest stable realease)
the rom i use is my dump so 100% original 
and i started playing like 1 week ago or so

profi200, 2022-08-30:
The reason for this is that the game expects to start at 0.0.2000 00:00 and to make RTC handling easier i force the RTC to the current date and time. Not exactly the way to handle it.

Not sure when i will fix this. There are only 2 options: Append the RTC offset to the save file or write it to a separate file.

FIFFY-Fede, 2022-08-30:
Ahh i understand, thanks a lot for the answer

FIFFY-Fede, 2023-02-03:
> Not sure when i will fix this. There are only 2 options: Append the RTC offset to the save file or write it to a separate file.

Honestly i think the separate file is the way to go to handle this. So maybe is compatible with putting save files in real cartridges. 

Using a tool like joey jr to dump or write save files from/to an original gba cartridge there are two files, the .sav and the .RTC
i don't know if this is the case for every game so for sure i'm wrong but it seems a nice solution to sync and use games on original hardware as well
Also i know that some games dont like having the rtc information on the same .sav but i also know so little over this topic

profi200, 2023-02-03:
If the emulator or whatever is implemented properly the extra bytes at the end should be ignored and cause no problem. Since mGBA also adopted the extra bytes in the save file i will probably go for this format.

hjk321, 2025-01-10:
Is there any way for a homebrew game to account for this on open_agb_firm in the meantime?

profi200, 2025-01-11:
Does that game handle time and date similar to Pokemon games? As in the game always resets to the current century on starting a new game. Essentially using the RTC just to count days passed since savegame start instead of for displaying the actual time.

luckytyphlosion, 2025-01-15:
It is a Pokémon ROMhack yes