profi200/open_agb_firm #83: Colors dim on default settings (9/15/22)

Bluekazoo, 2022-09-15:
Need help with Open agb firm brightness and colors. Games look dim even with backlight set to 142.

My current settings in the config.ini located in SD/3ds/open_agb_firm/ is

[General]
backlight=142
directBoot=false
useGbaDb=true

[video]
scaler=2
gbaGamma=2.2
lcdGamma=1.54
contrast=1.0
brightness=0.0

[advanced]
saveOverride=false
defaultSave=14

Setting backlight to 142 makes things brighter but the colors are still dim. Changing contrast and brightness causes things to become distorted and glitchy.
lower gba gamma but higher Lcd gamma makes the game washed out bad. The inverse makes things darker.

I'm using the current release (open_agb_firm_alpha_20220425) on N3DS.

Top image is openabgfirm and the bottom is an inject.
[image]

[image]

Example of glitchy-
[image]

profi200, 2022-09-18:
Try the latest build from the "Actions" tab above. And set lcdGamma to 2.2 (disables gamma correction).

mirrorbender, 2024-01-07:
I'm on and o2ds and still experiencing what I think is the same issue. I'm running the 10/6/2023 release, and the max brightness is noticeably lower than VC injects or the 3ds menu. Brightness is set to 117 (max for o2ds) in config, and control via x+up/down works, but the screen just doesn't get as bright. I have set the lcdGamma to 2.2, that is not the issue.

I have also tried directly booting open_agb_firm as boot.firm to see if maybe the issue was due to booting through the luma chainloader, but that seemed to make no difference.

profi200, 2024-01-07:
Backlight brightness is currently limited for safety. I can't allow more because there are too many different LCDs with different brightness curves. If i allow too much this can damage the backlight or the PMIC for specific display types. Only with calibration i can allow more. That is in the works.