profi200/open_agb_firm #86: Can't adjust the backlight or take screen dumps

celloh123, 2022-10-11:
Pressing X+UP, X+DOWN, X+LEFT or X+RIGHT does nothing and pressing SELECT + Y also does nothing

profi200, 2022-10-11:
Use the latest build from the Actions tab above.

profi200, 2022-10-17:
I don't quite understand your wording. I can't skip any of the steps because the frame coming from the hardware is in a different orientation than what the physical LCDs on 3DS use. So rotation is necessary. The GPU is also unfortunately tile based so all textures processed by the GPU are swizzled which makes the display transfer at the end necessary to unswizzle.

Nxt3, 2023-02-17:
@profi200 The build artifact expired; is there any way we could get another build there?

Thanks!

profi200, 2023-02-17:
Done.

Nxt3, 2023-02-17:
Thank you so much!!