profi200/open_agb_firm #91: Can you please implement manual enable of headphone output?

kylxbn, 2022-12-28:
Since there is an unsolved bug where some headphones are not recognized when plugged in and audio continues to play via the built-in speakers, can we please have a config setting (that will temporarily enable headphone output) or a key shortcut (that will toggle between speaker and headphone outputs)?

profi200, 2022-12-31:
Fine, i'm gonna add an override. If you have one of these headphones with buttons try to hold the buttons for a few seconds.

celloh123, 2023-02-16:
> Fine, i'm gonna add an override. If you have one of these headphones with buttons try to hold the buttons for a few seconds.

is there no other way? my headphones dont have buttons.

profi200, 2023-02-17:
Other than modifying the code, no.

profi200, 2023-08-05:
This is now implemented. See nightlies/actions tab.

closed 2023-08-05 by profi200