profi200/open_agb_firm #92: Maybe remove the "Sound has lots of aliasing issues" limitation?

kylxbn, 2022-12-31:
The phrase makes it sound like the 3DS has a 3DS-hardware specific bug that makes the audio have lots of aliasing compared to an actual Game Boy Advance. I will admit that I compared using a GBA emulator (mGBA) because I don't have a GBA, but the audio from open_agb_firm (or rather, the internal GBA hardware in the 3DS) sounds exactly the same compared to a GBA with all its aliasing. I compared with both a spectrum chart and my ears.

[image]
[image]

The GBA actually has an 8-bit DAC (FIFO) and that's just how it actually sounds--very fuzzy and hissing. To make things worse, some (many?) GBA games have mixing bugs that make the audio sound worse than it could actually be.

I would like to propose that the "Sound has lots of aliasing issues" limitation be removed because that hardware limitation is actually how the GBA works so it should be a feature instead of a limitation. Unless, of course, I am misunderstanding something and the 3DS actually has a hardware bug that makes the audio worse than how an actual GBA sounds like.

profi200, 2022-12-31:
Well, compare against a real GBA. Or these: https://gc-forever.com/wiki/index.php?title=Game_Boy_Interface/Standard_Edition#Sound_emulation

kylxbn, 2023-01-13:
Hi, @profi200 , thank you very much for the kind reply! Thank you for explaining the difference between how the GBA and 3DS outputs audio--I always kept in mind the probability that I was misunderstanding something because I am not familiar with either audio hardware.

Most importantly, thank you very much for sharing hardware audio recordings of the GBA! I tried comparing both systems using the Pokemon Sapphire GBA recording. I did not change anything from the hardware GBA recording except normalize the DC offset (using a high-pass filter at 20Hz), adjust/match volume, and resample to 48,000 Hz--these changes should not introduce any audible difference. As for my 3DS recording, I used a Behringer UCA222 which has sufficiently good-enough noise floor.

Here are spectrum screenshots:

GBA:
[image]

New 3DS LL:
[image]

Again, I am not trying to be arrogant or anything--I literally don't know anything about either audio hardware. But looking at the screenshots, it looks like the GBA has a higher noise floor (around 6dB more starting from 6000Hz). Also, the 3DS seems to have a sample rate of 34,700Hz. The GBA also has frequency spikes on random places (like at 2,091Hz and 4,192Hz (which is interestingly really close to 2,091*2 so it's maybe a harmonic?)) 

Listening to both using my Philips SHP9500 (which has relatively good frequency response), I hear that the 3DS has an ever-so-slightly muffled audio at the higher frequencies. The GBA has an ever-so-slightly more bright (noisy? alias-y?) sound. However, there are also times when there was virtually no difference to both audio. The PSG channels and FIFO channels match closely between each other, or they are too similar to hear a difference.

(Reference: Philips SHP-9500 frequency response)
[image]

My (admittedly possibly misinformed and wrong) conclusion is that, it seems that just like you said, the GBA outputs audio via PWM which introduces more aliasing to the audio on top of the very noisy 8-bit bit depth output. While the 3DS outputs audio using a proper DAC so that no extra aliasing is added, which results in less noise (which makes the audio sound a tiny bit more muffled compared to a GBA).

In that case, personally, I don't think it should be considered a limitation in `open_agb_firm` nor the 3DS--just a hardware difference that is probably not really audible in the end unless compared side by side. But of course, in the end, I will leave the decision to you! This was just a suggestion, and either way, I can't be more thankful for this project and your contribution regardless of its limitations. Thank you very much for `open_agb_firm`!

Audio files I compared: Download [here](https://mega.nz/folder/cBNjmYyD#X7777QComsF7H-KOvBppig)

profi200, 2023-01-13:
I don't know if the spectrum analyzer really can show this but if you compare original GBA hardware vs. 3DS the output from 3DS sounds sort of like an old telephone/highly compressed audio/more "crisp" (the last is what many people told me). The aliasing is somehow reduced a bit by the filtering on a real GBA but on 3DS it stands out much more. It's hard to describe.

vpelletier, 2023-05-14:
How were the samples captured ?

To my untrained ears, and just listening to the boot chime (because it is so easy to get to repeatedly), there are significant differences between headphones and the built-in speaker, both on the same GBA.

To me, GBA with headphones sounds similar to open_agb_firm with headphones, while the GBA's built-in speaker (and possibly the rest of the circuit after it splits away from the jack output) hides the "grittiness" of the high-pitched notes at the end of the boot chime.

I suspect the vast majority of GBA users are used to the sound of the built-in speaker. Which in turns means that the higher-fidelity (?) built-in speakers of the 3DS could be what causes the difference.

This said, I think I can also hear a difference between the 3DS speakers and headphones, to me the speakers have a more pleasant sound for mid-range frequencies.

Test hardware:
- Audio-technica ATH-SX1a studio monitor headphones
- GBA AGB-CPU-02 and AGB-CPU-03 (I could not hear a difference between these models)
- New 3DS LL

And technically also the GC Game Boy Player, but this one's signal went through more devices than I care to list.

profi200, 2023-05-14:
That's a point i had not considered regarding speaker vs. headphones on a real GBA. I rarely use headphones and GBA speaker output sounds better than 3DS speaker output to me.

The filtering used on GBA for the speaker (and headphone output?) is apparently inverse butterworth with 10 kHz cutoff. On 3DS there is only very mild filtering and it's mostly digital filtering. I could not figure out fully what their filters do yet but it mostly looks like phase correction/normalization to fix the bad frequency response of the crappy, small speakers.

kylxbn, 2023-05-14:
I apologize if my original comments were hard to understand. Since the very beginning, I was comparing the output of both the GBA and my N3DSLL from the digital perspective. I was completely disregarding the quality of either speakers or what (analog) filters were applied before output because if we were to put the built-in speakers for consideration, there there will be lots of variables to confirm, including but not limited to the type, size, and overall quality of the speakers.

All the tests and listening I did were performed by recording the headphone output of the 3DS (and using @profi200 's [hardware GBA audio recording links](https://gc-forever.com/wiki/index.php?title=Game_Boy_Interface/Standard_Edition#Sound_emulation) and comparing both by playing the recordings and alternating between both on my computer.

With that method, I admit there is a difference but in my opinion, the difference is too insignificant that I can't tell if it's the way the audio is rendered by the DAC or if it is just the audio circuitry that makes the difference. That is why I proposed that the "Sound has lots of aliasing" issue be considered a non-issue because it is not some sort of bug that significantly alters the sound playback capability of GBA games in 3DS hardware. It's just how the audio circuitry differs on both systems.

For example, I would probably consider it a bug if the audio data written by the game is least-significant-first and is properly understood by GBA hardware but 3DS hardware interprets it as most-significant-first—that would definitely involve *an extreme amount* of distortion. Or, GBA audio samples are 8-bits wide but the 3DS ignores the 4 least significant bits and instead just plays 4-bit samples which would indeed in noticeable aliasing.

For anyone wanting to listen to my audio recordings, I have fixed the deleted link so you can download the WAV files and compare them yourself. They are available [here](https://drive.proton.me/urls/HHVF74ZW5G#M0s9rn39ZNmu).

Again, I don't really know what I'm talking about. I'm not as familiar with hardware like @profi200 :) So take my suggestion lightly.

profi200, 2023-09-26:
A little bit of an update. I think we found a very good example that proves my point about the aliasing. Run "Mother 3 (Japan)", go to Sound Player and select track 12. And compare this between GBA/DS lite and 3DS. You will immediately notice the aliasing. The going theory is that the sound has been down sampled too low in hardware.

Unfortunately i don't have any device with line in capability to do proper recordings (ideally with 192 kHz or something to prevent further losses). If anyone knows a good USB audio capture card that does not break the bank but delivers high quality let me know.

Here are 2 recordings (arguably with a bit too low sample rate). One is from a DS lite and the other from a N3DS XL. If you compare to a GBA also keep in mind the GBA runs ever so slightly faster than DS and 3DS so expect a small pitch change.

Thanks for beta215 for doing these recordings.
[Mother_3_audio_aliasing_test.zip](https://github.com/profi200/open_agb_firm/files/12729937/Mother_3_audio_aliasing_test.zip)

vpelletier, 2023-09-29:
While I am firmly on the sidelines here, I cannot help but be pulled back by this topic.

So I dug a bit more, and found this homebrew: [gba-sound-demo](https://github.com/velipso/gba-sound-demo). This is essentially a sine signal generator for the GBA. While the main goal of this project was visibly to explore how (or prove it was possible ?) to make high-bitrate audio on the GBA, I consider this a perfect testing tool for sound quality. I do not have a GBA flashcart, so I cannot test it on real hardware, but I can of course test on 3DS.

The testing equipment is... my android phone (pixel 5), with an [open-source oscilloscope app with spectrum display](https://f-droid.org/en/packages/org.billthefarmer.scope/). Admittedly primitive and totally not calibrated, but this *should* mostly matter on the volume axis, but I expect something at least decent on the frequency axis.

The following screenshots are taken with the GBA (...3DS, really) at a sample rate of 32k, 8-bits samples, no dithering and 32k hardware sample rate, and some constant frequency around 500Hz.

Speaker output, volume slider at around 40%:
[image]

Speaker output, volume slider at around 70%:
[image]

Headphone output with my phone stuck accross the headset, volume slider at 100%:
[image]

The low-frequencies in the first screenshot must be ambient noise, at such low-ish sound level it would still be picked up in addition to the signal.

As you can see (and as I could clearly hear), there are significant harmonics when going through the built-in speakers and "pushing" the volume. Those do not appear with the headphone output - at least not with this setup - and the volume with headset on would have been painfully loud.

Does it prove anything ? Well, the 3DS speakers are not great (who saw that coming ?) and introduce a lot of their own resonances when enough energy is pumped into them. Beyond that, I'm no sound engineer, all I can say is that the spectrogram captured through the headphones looks clean: I do not see signs of the signal being output is anything other than a sine.

Playing some more with the headphones setup, I see the a strong low-pass filter kicking in above 12kHz, by 14kHz there is very little sound being picked up. It could be my headphones (the same studio monitors that I used in my previous tests) or the phone though... By my ears, the sound intensity does drop a lot and consistently with the spectrogram, but then again, mark-I ears and all that.

Maybe someone can make better use of this homebrew rom and I could.

profi200, 2023-09-29:
I would try this. It generates a square wave and you can adjust the frequency. When changing frequencies you can already hear the aliasing. Can't get more clear than this:

There is also a FIRM included that plays sound natively and you immediately hear how much cleaner that is.
[audio_test.zip](https://github.com/profi200/open_agb_firm/files/12767890/audio_test.zip)

profi200, 2024-08-30:
I think it's 2 problems here. The mentioned aliasing which is a hardware bug and the different frequency response of the speakers that makes the unwanted noise/aliasing sound louder than it is. To some extend that noise is also present on a real GBA but there it's almost all filtered out.