profi200/open_agb_firm #99: How do I actually use the per game ini feature?

Adamillo, 2023-03-05:
I've tried to use the game ini feature based on what was explained in the readme on the repo, but I could not get it to work. Here's the path I put it on and here's what the contents of the ini looks like
[image]
[image]

profi200, 2023-03-05:
Have you tried the latest build from the "Action" tab above?

Adamillo, 2023-03-05:
I did use the latest build in order to be able to even change the brightness

profi200, 2023-03-05:
Ah, i only saw it now. Remove the "u8 " in both lines.

Adamillo, 2023-03-05:
Ah, I see why I made that mistake. The u8 in those options should be removed from the readme in the repo to avoid confusion