profi200/open_agb_firm #103: [Question] OAF Power consumption

Bebop182, 2023-03-21:
How awake is the 3DS system hardware when running open_agb_firm? At first I thought that only the GBA hardware might be running but the inputs and displays most likely aren't managed by the GBA hw. So I suppose the 3DS probably runs normally in the background. Is there potential for power consumption improvement or are there hard limitations?

Thank you

Masamune3210, 2023-03-21:
Most of the 3DS hardware is turned off while in TWL or AGB mode.

Masamune3210, 2023-03-21:
TWL and AGB read and write to system components through the firms loaded on the ARM9 iirc, it's the one that arbitrates access to hardware for NATIVE i.e. 3DS mode anyway

Bebop182, 2023-03-21:
Thanks a lot for your answer. I noticed that the file explorer seems to run on the main arm 11 cpu. So after selecting a game the console would switch to the GBA hw and the main 3DS cpu would go to sleep?

Edit: As I'm reading through the code it seems that most of oaf runs on the main cpu even while the game is running. For example the updateBacklight static method which check for a key combo to adjust brightness is probably running every frame in oafupdate, or actually maybe completely decoupled from the GBA rendering.

I'm used to do programming at a much higher level, quite far from hardware. Could you recommend resources that could help me understand better how is open_agb_firm hooked and what the limitations are? Like why does it need to run that backlight code on the main cpu and would it be possible to put either arm11 or arm9 cpu to sleep?

I really would like to be able to contribute in the future

Masamune3210, 2023-03-21:
I will leave that for the actual devs to answer, I have not really messed with the 3DS in a programming sense much