profi200/open_agb_firm #109: [Feature Request] Option to Force Audio through Speakers

Zarathestrio, 2023-04-27:
Ahoy! I've recently run into an issue where my N3DS constantly thinks that a pair of headphones are plugged into the jack. While this could be theoretically fixed on the hardware side, I've been looking for a software fix. Luckily the guys over at Luma3DS have already made a custom boot file to grant the option of forcing all audio through the speakers, but it doesn't work for anything that doesn't work directly through Luma. I was wondering if there was any possibility that a feature could be added to allow a similar feature for open_agb_firm. Any help is greatly appreciated!

profi200, 2023-05-06:
Same as #91 but speakers.

profi200, 2023-08-05:
This is now implemented. See latest commit/actions tab.

closed 2023-08-05 by profi200