profi200/open_agb_firm #110: [Feature Request] Color Correction

Memo-Ri, 2023-05-06:
Greetings! Would like to kindly request for an option to apply color correction, or a setting that would allow to edit/change the color tone of the display. Would like to have colors like the ones here: https://twitter.com/Lewchube/status/1541758514930094081
I tried playing around with the gamma values but it doesn't seem to allow for it. Attached a picture as well for your convenience. 
[image]

Thank you so much for working on this project!

profi200, 2023-05-06:
Can't find where but pretty sure i explained it somewhere. The situation is the GPU lacks the feature (fragment shaders) to implement this accurately. So i'm left with very few options to implement this. Can't promise that i  will be able to implement it.

Squall-Leonhart, 2025-03-21:
closed by https://github.com/profi200/open_agb_firm/commit/5356e5c89a8f85b708fa498a2ead70b3a8368021  ???