profi200/open_agb_firm #111: GBADoom White Screen

RetroGamer02, 2023-05-11:
Hi, I noticed that DoomHacks GBADoom does not run and just displays a white screen.
I know there is plenty of other ways to play Doom but If possible it would be nice to be able to play this on 3DS.
If its not possible or not worth doing no worries I still have my DS Fat. =)

profi200, 2023-05-14:
Which save type does this game use? SRAM? You could try the latest nightly build at the actions tab above and override the save type. Maybe the game will start working.

RetroGamer02, 2023-05-14:
It does use SRAM but I tried the override and no luck.

profi200, 2023-05-14:
The reason the game is freezing is it uses the fastest possible ROM wait states which is not supported by the hardware. ROM wait cycles on first access must be 3 but this game sets it to 2.
https://problemkaputt.de/gbatek-gba-system-control.htm

You can ask the dev to add detection for slow flashcarts/hardware similar to how OpenLara did it here: https://github.com/XProger/OpenLara/blob/master/src/platform/gba/checkROM.cpp

As a temporary workaround open the ROM in a hex editor, go to offset 0x7E20 and patch bytes DA46 to B645. The game runs fine but it may be slower than intended (looked fine for me).

RetroGamer02, 2023-05-14:
> 
Will give it a try THANKS! =)

closed 2023-05-14 by RetroGamer02