profi200/open_agb_firm #113: Error while booting open_agb_firm

crystalstorm466, 2023-05-17:
Model: New 3DS XL (US)

When ever I load a rom (currently only tested Pokemon Sapphire) the system would display two error messages and hang.

The two error messages are in order:
`Could not access gba_db.bin! Press any button to continue.`
`Errors: fatfs file not found.`

After these error messages the system would hang for an unspecified amount of time and would only show the gba splash screen then booting when I would press the power button. 

Picture is included below:
[image]

profi200, 2023-05-17:
Either you didn't copy the included gba_db.bin to the SD card or your SD card filesystem is corrupt.

closed 2023-05-17 by crystalstorm466

profi200, 2023-05-17:
Extract the "3ds" folder inside the archive to the root of the SD card. If it asks if you want to merge folders/files then click yes.